node-usb/node-usb #537: FATAL ERROR:  Polling is not active

cleoo, 2022-08-31:
Hello,

How to support hot unplugging a USB device?
As soon as I use the startPoll() function, the application crashes with the error indicated in the title. The endpoint used in the code is an Interrupt IN endpoint of a Cdc Acm communication interface.

## Code
```
const usb = require('usb')
const dev = usb.findByIds(0x05f9, 0x4204)

dev.open()

const _if = dev.interface(0)
if (_if.isKernelDriverActive()) _if.detachKernelDriver()
_if.claim()

const _ci = _if.endpoint(0x81)

_ci.startPoll()

_ci.on('data',  (...args) => console.log('data',  ...args))
_ci.on('end',   (...args) => console.log('end',   ...args))
_ci.on('error', (...args) => console.log('error', ...args))
```

## Steps to Reproduce:
1. Plug device
2. Start program
3. Unplug device

## Output
```
error [Error: LIBUSB_TRANSFER_NO_DEVICE] { errno: 5 }
error [Error: LIBUSB_TRANSFER_NO_DEVICE] { errno: 5 }
FATAL ERROR:  Polling is not active.
 1: 0xb09c10 node::Abort() [node]
 2: 0xa1c193 node::FatalError(char const*, char const*) [node]
 3: 0xa1c19c  [node]
 4: 0xad1abb napi_fatal_error [node]
 5: 0x7f28c8213006  [/home/[...]/node_modules/usb/prebuilds/linux-x64/node.napi.glibc.node]
 6: 0x7f28c822193b handleCompletion(Transfer*) [/home/[...]/node_modules/usb/prebuilds/linux-x64/node.napi.glibc.node]
 7: 0x7f28c821333d Napi::ThreadSafeFunction::CallJS(napi_env__*, napi_value__*, void*, void*) [/home/[...]/node_modules/usb/prebuilds/linux-x64/node.napi.glibc.node]
 8: 0xacfc61  [node]
 9: 0x1564956  [node]
10: 0x1577084  [node]
11: 0x15652a8 uv_run [node]
12: 0xa43dd5 node::SpinEventLoop(node::Environment*) [node]
13: 0xb4bdb6 node::NodeMainInstance::Run(node::EnvSerializeInfo const*) [node]
14: 0xacd592 node::Start(int, char**) [node]
15: 0x7f28cad542d0  [/usr/lib/libc.so.6]
16: 0x7f28cad5438a __libc_start_main [/usr/lib/libc.so.6]
17: 0xa4067c  [node]
Abandon (core dumped)

```

## Additional Information
- Operating System: Archlinux
- Runtime v16.15.1
- Node USB Version 2.5.1

thegecko, 2022-08-31:
Please use the [usb events](https://github.com/node-usb/node-usb#usbdetection) for hotplug support. e.g.:

```typescript
const usb = require('usb').usb;

const attach = device => console.log('attach', device);
const detach = device => console.log('detach', device);

usb.on('attach', attach);
usb.on('detach', detach);
```

Endpoint polling is something different.

cleoo, 2022-08-31:
Merci pour votre réponse. J'ai dû mal me faire comprendre. L'event "detach" n'a pas le temps d'être appelé, le programme a planté avant.

thegecko, 2022-08-31:
OK, I'm not familiar with the endpoint polling mechanism, this will need investigating.

PatrickHallek, 2022-09-21:
I am facing the same issue.
- Operating System: Archlinux (aarch64)
- Runtime v16.13.2
- Node USB Version 2.5.0

```shell
2022-09-21 15:08:39: FATAL ERROR:  Polling is not active.
2022-09-21 15:08:39:  1: 0xaf4198 node::Abort() [node]
2022-09-21 15:08:39:  2: 0xa1ea88 node::FatalError(char const*, char const*) [node]
2022-09-21 15:08:39:  3: 0xa1ea94 void node::FPrintF<>(_IO_FILE*, char const*) [node]
2022-09-21 15:08:39:  4: 0xac0d84 napi_fatal_error [node]
2022-09-21 15:08:39:  5: 0x7f703c1458  [/home/pi/development/backend/node_modules/usb/prebuilds/linux-arm64/node.napi.armv8.node]
2022-09-21 15:08:39:  6: 0x7f703cec40 handleCompletion(Transfer*) [/home/pi/development/backend/node_modules/usb/prebuilds/linux-arm64/node.napi.armv8.node]
2022-09-21 15:08:39:  7: 0x7f703c17ec Napi::ThreadSafeFunction::CallJS(napi_env__*, napi_value__*, void*, void*) [/home/pi/development/backend/node_modules/usb/prebuilds/linux-arm64/node.napi.armv8.node]
2022-09-21 15:08:39:  8: 0xabf40c  [node]
2022-09-21 15:08:39:  9: 0x14bdbe4  [node]
2022-09-21 15:08:39: 10: 0x14ced20  [node]
2022-09-21 15:08:39: 11: 0x14be558 uv_run [node]
2022-09-21 15:08:39: 12: 0xa44a14 node::SpinEventLoop(node::Environment*) [node]
2022-09-21 15:08:39: 13: 0xb2f620 node::NodeMainInstance::Run(node::EnvSerializeInfo const*) [node]
2022-09-21 15:08:39: 14: 0xabd198 node::Start(int, char**) [node]
2022-09-21 15:08:39: 15: 0x7f9180ae18 __libc_start_main [/lib/aarch64-linux-gnu/libc.so.6]
2022-09-21 15:08:39: 16: 0xa42c3c  [node]
2022-09-21 15:08:42: Aborted (core dumped)
```

cleoo, 2022-09-22:
hello, 
To avoid the problem I simply replaced the use of usb polling by an implementation based on a Stream:Readable continuously invoking the transfer method of the corresponding endpoint.

```js
const { Readable }  = require('node:stream')
// ...
const ep = ...
const stream = new Readable({
      read(size) {
        ep.transfer(
          Math.min(size, ep.descriptor.wMaxPacketSize), 
          (error, data) => {
            if (error)  this.emit('error', error)
            else        this.push(data)
          }
        )  
      }
```

PatrickHallek, 2022-09-26:
@cleoo 

This solution does not fix it for me. I am getting another error with the same stack trace:
``` bash
FATAL ERROR:  LIBUSB_TRANSFER_STALL
 1: 0xaf4198 node::Abort() [node]
 2: 0xa1ea88 node::FatalError(char const*, char const*) [node]
 ...
```

thegecko, 2022-10-31:
It looks like this could be caused by the polling not stopping properly when a device is detached.

This looks similar to https://github.com/node-usb/node-usb/issues/548 and may be resolved with the proposed fix there.

strazzere, 2022-10-31:
I've never seen the STALL, however #548 should fix the cascading `Polling is not active.` when many threads are used and an unplug event occurs.

strazzere, 2022-12-06:
I'm not following what you're asking. With the latest code, are you able to reproduce aborts being thrown, meaning their is another issue at hand?

Or are you just asking, fundamentally, it worries you that this had occurred?

PatrickHallek, 2022-12-07:
> Or are you just asking, fundamentally, it worries you that this had occurred?

Yes I do, your fix works fine (because it avoids calling `stopPolling` multiple times) , but I am not understanding why an error throwing leads to a FATAL ERROR that crashes the complete application. This behaviour not only takes place for this specific error but all errors that are getting thrown in the endpoint implementations. There seems to be a bigger issue in the error handling in general.

```js
public stopPoll(callback?: () => void): void {
        if (!this.pollActive) {
            throw new Error('Polling is not active.'); // Why does this leads to an unhandled FATAL ERROR that crashes node
        }
        // ...
}
```

strazzere, 2022-12-08:
The error leads to a fatal error because there is no one expecting it or catching it, so it bubbles up to the main thread. It is being raised via `transferDone` as a callback.

You are correct that something was wrong, though it is now fixed and the error would not be thrown anymore as this case can no longer occur.

Maybe this is just an opinion though, no idea what you are expecting the library to do. If it has another bug like this, then yes, it would throw an error erroneously again like it did. :shrug:

thegecko, 2022-12-10:
> @thegecko Can you please publish a release containing this fix?

@PatrickHallek this is now in the `v2.6.0` release.

Apologies this took a while, I had to upgrade the CI pipeline

closed 2023-02-26 by thegecko