node-usb/node-usb #538: Transfet out throws error

rychtaradam, 2022-09-02:
## Bug Description:
I've got a problem with sending a transferOut to my USB device in my Electron application.

Here is the code where the problem occurs:

```
async sendDataToDevice(device) {
        const data = new Uint16Array(16);
        data[0] = 3333;
        
        await device.transferOut(0x81, data);
}
```

this code throws for me not yet known reason this error:
`Error: transferOut error: TypeError [ERR_INVALID_ARG_TYPE]: The "size" argument must be of type number. Received an instance of Buffer`

## Additional Information
- Operating System: Windows 10
- Runtime (`node -v` or `electron -v`): v14.16.1
- Node USB Version (`npm ls usb`): 2.4.3

mildsunrise, 2022-09-03:
hmm, we have some bugs because of the design of the legacy API which mostly affect races and errors.
endpoint 0x81 is an input endpoint, right? in USB, the MSB is set for input endpoints AFAIK, so you're trying to do an output transfer to an input endpoint.
in any case we can leave this open since the error should be more intuitive / up to spec.

rychtaradam, 2022-09-05:
Sure, even if I use 0x02 which is my configured OUT endpoint on my device (I'm sure this is the right one), it then throws `Error: transferOut error: Error: LIBUSB_ERROR_IO`. 

Here is the USBView report:
```
Is Port User Connectable:         yes
Is Port Debug Capable:            no
Companion Port Number:            0
Companion Hub Symbolic Link Name: 
Protocols Supported:
 USB 1.1:                         yes
 USB 2.0:                         yes
 USB 3.0:                         no

Device Power State:               PowerDeviceD0

       ---===>Device Information<===---
English product name: "TEST DEVICE 001"

ConnectionStatus:                  
Current Config Value:              0x01  -> Device Bus Speed: Full (is not SuperSpeed or higher capable)
Device Address:                    0x31
Open Pipes:                           2

          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0200
bDeviceClass:                      0x00  -> This is an Interface Class Defined Device
bDeviceSubClass:                   0x00
bDeviceProtocol:                   0x00
bMaxPacketSize0:                   0x08 = (8) Bytes
idVendor:                        0x04D8 = Microchip Technology Inc.
idProduct:                       0xEA39
bcdDevice:                       0x0001
iManufacturer:                     0x01
     English (United States)  "MECA-SIM-HARDWARE s.r.o."
iProduct:                          0x02
     English (United States)  "TEST DEVICE 001"
iSerialNumber:                     0x00
bNumConfigurations:                0x01

          ---===>Open Pipes<===---

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0008 = 0x08 bytes
bInterval:                         0x01

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x02  -> Direction: OUT - EndpointID: 2
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0008 = 0x08 bytes
bInterval:                         0x01

       ---===>Full Configuration Descriptor<===---

          ===>Configuration Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x02
wTotalLength:                    0x0029  -> Validated
bNumInterfaces:                    0x01
bConfigurationValue:               0x01
iConfiguration:                    0x02
     English (United States)  "TEST DEVICE 001"
bmAttributes:                      0x80  -> Bus Powered
MaxPower:                          0x32 = 100 mA

          ===>Interface Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x04
bInterfaceNumber:                  0x00
bAlternateSetting:                 0x00
bNumEndpoints:                     0x02
bInterfaceClass:                   0x03  -> HID Interface Class
bInterfaceSubClass:                0x00
bInterfaceProtocol:                0x00
iInterface:                        0x02
     English (United States)  "TEST DEVICE 001"

          ===>HID Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x21
bcdHID:                          0x0101
bCountryCode:                      0x00
bNumDescriptors:                   0x01
bDescriptorType:                   0x22 (Report Descriptor)
wDescriptorLength:               0x001F

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0008 = 0x08 bytes
bInterval:                         0x01

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x02  -> Direction: OUT - EndpointID: 2
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0008 = 0x08 bytes

```

And here is the device descriptor in case it might be helpful:
```
0x05, 0x01, // USAGE_PAGE (Generic Desktop)
    0x09, 0x04, // USAGE (Joystick)
    0xA1, 0x01, // COLLECTION (Application)
        0x09, 0x01, // USAGE (Pointer)
        0xA1, 0x00, //Collection (Physical)
            0x05, 0x01, // USAGE_PAGE (Generic Desktop)
            0x09, 0x33, // USAGE (Rx)
            0x09, 0x34, // USAGE (Ry)
            0x09, 0x35, // USAGE (Rz)
 
            0x15, 0x00, // LOGICAL_MINIMUM (0)
            0x26, 0x00, 0x10, // LOGICAL_MAXIMUM (65535)
 
            0x75, 0x10, // REPORT_SIZE (16)
            0x95, 0x03, // REPORT_COUNT (3)   
            0x81, 0x02, //INPUT (Data,Var,Abs)
 
        0xC0, // END_COLLECTION
    0xC0 //END_COLLECTION
```

closed 2022-09-08 by rychtaradam