node-usb/node-usb #539: Methode transferIn freezes in node using NFC Reader ACR122u.

jarda51, 2022-09-02:
I have been trying to write a routine in node.js which should read UID of an inserted card in NFC Reader ACR122u. I can power up the reader, I get ATR if there is a card, but when I try to use the transferIn method it freezes. See the code:

```
import { usb, WebUSBDevice, findByIds } from 'usb';

const deviceU= findByIds( 0x072F, 0x2200);  // Find usb device by ID => ACR122u

const device = await WebUSBDevice.createInstance(deviceU); // turning usb into WebUSB
await device.open(); 
let config = await device.selectConfiguration(1);                           
await device.claimInterface(0);                // claim interface

let powerUpDevice  = new Uint8Array([0x62,0x00, 0x00, 0x00, 0x00,0x00,0x00,0x01,0x00, 0x00]).buffer;
let powerUp = await device.transferOut(2, powerUpDevice); // device ON => o.k.

let result = await device.transferIn(2, 64); // get ATR => o.k.
console.log(result.data.buffer);

// according to the vendor docu 
//Example:
//1. To get the serial number of the “connected PICC”
//UINT8 GET_UID[5]={0xFFh, 0xCAh, 0x00h, 0x00h, 0x04h};

let result2 = await device.controlTransferOut({
  'requestType': 'class',
  'recipient': 'interface',
  'request': 0xCA, 
  'value': 0x0000,
  'index': 0x02});

  console.log(result2); // transfer o.k.

  let answer = await device.transferIn(2, 64); // here !!!!! it stucks !!!!! <============
  console.log(answer);

```

## Additional Information
- Operating System: Windows 10 , I´m using WinUSB driver providing by Zadig 
- Runtime (`node -v` or `electron -v`):  v14.17.6
- Node USB Version (`npm ls usb`): usb@2.5.1

mildsunrise, 2022-09-03:
if I remember correctly, WebUSB does some kind of busy wait for the locking? this is something I wanted to change, but I don't know its current state.

@jarda51 can you test using the legacy API instead of the WebUSB adaptor?

jarda51, 2022-09-03:
I´ve tried the legacy API, but I´m not sure about the right methods. What is the device.controlTransferOut from WebUSB in the legacy API? Can you give me a hint? I tried something like this:

`import usbLib, { usb, WebUSBDevice } from 'usb';

const device= usbLib.findByIds( 0x072F, 0x2200);  // Find usb device by ID
await device.open();                                // open device
await device.interfaces[0].claim();                    // claim interface

 const endpoint    = device.interfaces[0].endpoints[0];
 const outEndpoint = device.interfaces[0].endpoints[1];
 const inEndpoint  = device.interfaces[0].endpoints[2];

 let powerUpDevice    = new Uint8Array([0x62,0x00, 0x00, 0x00, 0x00,0x00,0x00,0x01,0x00, 0x00]).buffer;
 await outEndpoint.transfer(powerUpDevice); // device on => o.k.

  inEndpoint.transfer(64, function (err, data) {
        if (data) {
              console.log(data); // Getting ATR => o.k.
        } else if (err) {
          console.log(err)  
        } else {
          cconsole.log(end)
        } 
      });

let getCardUID = new Uint8Array([0xff,0xca, 0x00, 0x00, 0x04]).buffer;
    outEndpoint.transfer(getCardUID, function (err, end) {
      if (err) {
        console.log(err)
      }
      console.log(end) // 5 => o.k.
    });

   inEndpoint.transfer(64, function (err, data) {
        if (data) {
          console.log(data); 
        } else if (err) {
          console.log(err)  
        } else {
          cconsole.log(end)
        } 
      });
`

but it never reaches the inside of the callback function  in  inEndpoint.transfer method. Can you show me my procedure from the WebUSB in the legacy API? 
thank you very much!

jarda51, 2022-09-04:
I converted my code to the legacy API, but it freezes in the inEndpoint.transfer method too. 
here is the code:

`import usbLib, { usb, WebUSBDevice } from 'usb';

const device= usbLib.findByIds( 0x072F, 0x2200);  // Find usb device by ID
await device.open();                                // open device
await device.interfaces[0].claim();                    // claim interface

 const endpoint    = device.interfaces[0].endpoints[0];
 const outEndpoint = device.interfaces[0].endpoints[1];
 const inEndpoint  = device.interfaces[0].endpoints[2];

 let powerUpDevice    = new Uint8Array([0x62,0x00, 0x00, 0x00, 0x00,0x00,0x00,0x01,0x00, 0x00]).buffer;
 await outEndpoint.transfer(powerUpDevice); // device on => o.k.

// according to the vendor docu 
//Example:
//1. To get the serial number of the “connected PICC”
//UINT8 GET_UID[5]={0xFFh, 0xCAh, 0x00h, 0x00h, 0x04h};

      device.controlTransfer(0xa2,0xca, 0x00, 0x00, 0x00, function(error, data) {
        if(error) {
          console.log(error);
        }
        else if(data){
           console.log(data.buffer);

      inEndpoint.transfer(64, function (err, data) {  // here it freezes
        if (data) {
          console.log(data);  // it doesn´t reach this point
        } else if (err) {
          console.log(err)   // it doesn´t reach this point
        } else {
          console.log(end) // it doesn´t reach this point
        } 
      });
        }
      });`

mildsunrise, 2022-09-04:
Then I have no clue and I believe this may be an OS/backend specific issue... sorry :(

closed 2022-09-10 by jarda51