node-usb/node-usb #540: Native code crashing

gcampbell-msft, 2022-09-07:
## Bug Description:
I recently upgraded to node-usb 2.5.*, and I started noticing that the native code was crashing my program when plugging in or unplugging hardware. I was able to debug it with VS and found it hit a call stack inside of PNP (plug and play) in native code. 

## Steps to Reproduce:
1. Create a program where `usb.on("attach")` and `usb.on("detach")` is used.
2. Run the program and disconnect or connect a board.
3. Notice the failure.

Additional info: 
I downgraded to 2.4.3 and the issue went away.
Exception:
[image]

Call stack:
[image]

## Additional Information
- Operating System: Windows 10
- Runtime (`node -v` or `electron -v`): 16.13.1, node.
- Node USB Version (`npm ls usb`): 2.5.1

thegecko, 2022-09-07:
Versions suggest this has been introduced with the Windows USB detection code in https://github.com/node-usb/node-usb/pull/524

mildsunrise, 2022-09-08:
the fact that it's when delivering a PNP event also strongly suggests this. @Julusian have you seen this previously in node-usb-detection?

Julusian, 2022-09-08:
No, not something I've seen there, but the code here was written differently and with a newer api.

I have a theory, with a hopeful fix #541

thegecko, 2022-09-09:
The fix from @Julusian has been released using the `alpha` tag and can be installed using:

```bash
npm i usb@alpha
```

Could you test this fixes your issue?

gcampbell-msft, 2022-09-12:
@thegecko Yes, I'll try and test today. Thanks!

gcampbell-msft, 2022-09-12:
I just tested and it did not fix the issue. 

Below are two more images of the callstack and the exception.
[image]
[image]

Julusian, 2022-09-25:
@gcampbell-msft any chance of a reproduction script?
I don't think this is going to be an easy one to figure out, as that stack trace is completely inside windows code, and there isnt really much setup/teardown of the events for things to be done wrong..

I am not able to reproduce any crashing with a minimal script:
```
const {usb} = require('.')

usb.on('attach', (a) => {
    console.log('attach', a)
})
usb.on('detach', (a) => {
    console.log('detach', a)
})
```

yooouuri, 2022-10-05:
https://gist.github.com/yooouuri/5140bd9c806a98952aed455aa383f130
A crash report.

Im getting a 

> ProcessPlugPlayEventCallback
> Fatal Error: EXCEPTION_ACCESS_VIOLATION_READ / 0xa31054a2

In my error handling service (Sentry)

[image]

[image]

gcampbell-msft, 2023-01-12:
@Julusian Has there possibly been any more investigation or updates on this? 

I haven't been able to create a reproduction script, but I wonder if @yooouuri's crash report would be helpful.

Thanks!

thegecko, 2023-03-06:
Investigating this issue further, it certainly has to do with the changes introduced in #524 and is on Windows only.

Not necessarily due to the events in Windows, but more to do with the dropping of the polling code.

For some strange reason (which I will investigate further when I get time), the devices delta emitted from [emitHotplugEvents](https://github.com/node-usb/node-usb/blob/master/tsc/usb/index.ts#L56) when triggered by [the attachIds event](https://github.com/node-usb/node-usb/blob/master/tsc/usb/index.ts#L104) fail to open. This is fine when using the polling method triggered from [pollHotplug](https://github.com/node-usb/node-usb/blob/master/tsc/usb/index.ts#L85) :/

As a workaround, I've introduced a setting to force the use of polling for hotplug events: https://github.com/node-usb/node-usb/pull/576

code used to reproduce this:

```typescript
import { usb, WebUSBDevice } from 'usb';

usb.pollHotplug = true;

const attach = async (device: usb.Device) => {
    console.log('attach', device);
    try {
        const webDevice = await WebUSBDevice.createInstance(device);
        console.log(webDevice);
    } catch(error) {
        console.log(error);
    }
};

const detach = (device: usb.Device) => console.log('detach', device);

usb.on('attach', attach);
usb.on('detach', detach);
```

gcampbell-msft, 2023-03-07:
@thegecko Thanks for the update! I'll make a note to try out this and see if updating our dependency would work with this workaround. 👍

thegecko, 2023-03-08:
OK, the cause of this problem is a race condition, thanks @arekzaluski!

For some reason, Windows is firing attached events into the JavaScript code and libusb can get the details, but the device isn't yet ready to write.

@Julusian is there a later Windows event we can use to indicate hotplug or can we wait for something to be ready in the native Windows code?

In the meantime, I've introduced a dirty delay over at #577 which fixes the issue for now.

Julusian, 2023-03-08:
> For some reason, Windows is firing attached events into the JavaScript code and libusb can get the details, but the device isn't yet ready to write.

Which is then causing a segfault in the original event handler? I cant get my head around how or why this would happen, but I guess if delaying the open fixes it then so be it.

> @Julusian is there a later Windows event we can use to indicate hotplug or can we wait for something to be ready in the native Windows code?

I don't know windows apis well enough to answer that. Nothing jumps out at me from a similar page, other than https://learn.microsoft.com/en-us/windows/win32/api/winuser/nf-winuser-registerdevicenotificationw, which would require pretty much completely rewriting that windows hotplug detection. Perhaps justified to see if it makes this issue go away

----

I'm glad you've figured out how to reproduce it though, Ill give it another try myself and see if I can wrap my head around what is happening.

Could we perhaps be accidentally be blocking the event callback thread while doing some nodejs side stuff? Perhaps triggering some thread safety issues with libusb entering the same apis from within the same process?

Julusian, 2023-03-09:
I still cannot reproduce this with x64 nodejs, but if I switch to x86 (I saw the arch in the screenshot from @yooouuri ), then attach causes a segfault before I can `console.log` anything

@gcampbell-msft which are you using when you can trigger this?

Julusian, 2023-03-09:
Digging into it not working at all on x86 nodejs on windows, and I found that even an empty notification callback that did nothing would cause it to crash. I have a fix for that, which looks like a mistake I made in the initial implementation, and not knowing enough about the quirks of interfacing with windows apis.  
I can't say for certain that it will resolve the overall issue here, but the fix will affect x64 too so it is possible that it could fix it.

thegecko, 2023-03-09:
> Which is then causing a segfault in the original event handler?

Ah, no. I haven't encountered that issue. I found that immediately trying to open a device after it has notified as being attached caused a LIBUSB error. Perhaps there are multiple issues at play here?

> I found that even an empty notification callback that did nothing would cause it to crash. I have a fix for that

Great, thanks! I'll try this out later

thegecko, 2023-03-09:
Thanks @Julusian 

Unfortunately this hasn't fixed the issue, but it doesn't hurt either, so I've merged it.

One interesting piece of information which may help others reproduce this is that I've only seen this issue on USB devices which have a storage HID interface and open an explorer window. Simpler USB devices like the test board we use for node-usb don't exhibit this behaviour.