node-usb/node-usb #545: Legacy device not turned into WebUSB compatible device

amochkin, 2022-09-21:
## Bug Description: Legacy device not turned into WebUSB compatible device

## Steps to Reproduce:
1. Legacy device is found:
```javascript
{
  busNumber: 1,
  deviceAddress: 2,
  deviceDescriptor: {
    bLength: 18,
    bDescriptorType: 1,
    bcdUSB: 528,
    bDeviceClass: 0,
    bDeviceSubClass: 0,
    bDeviceProtocol: 0,
    bMaxPacketSize0: 64,
    idVendor: 6268,
    idProduct: 1360,
    bcdDevice: 512,
    iManufacturer: 1,
    iProduct: 2,
    iSerialNumber: 3,
    bNumConfigurations: 1
  },
  portNumbers: [ 4 ]
}
```
2. Turning it into WebUSB device:
```typescript
await WebUSBDevice.createInstance(legacyDevice);
```
3. `undefined` is returned

## Additional Information
- Operating System: Ubuntu 22
- Runtime (`node -v` or `electron -v`): node v14.19.3
- Node USB Version (`npm ls usb`): usb@2.5.1

mildsunrise, 2022-09-21:
could you post a small script we could run to reproduce the problem?

thegecko, 2022-09-21:
Failure to create a WebUSB device often happens when there are errors getting string data from the device.

We have a `try...catch` which is masking any error in this area we should probably remove:

https://github.com/node-usb/node-usb/blob/master/tsc/webusb/webusb-device.ts#L15

amochkin, 2022-09-21:
> could you post a small script we could run to reproduce the problem?

I think it is pretty much device specific. I am testing on a real device which is Alienware FX controlller, I am not sure my script will reproduce the problem without the device.

amochkin, 2022-09-21:
@thegecko 
> Failure to create a WebUSB device often happens when there are errors getting string data from the device. 
> We have a `try...catch` which is masking any error in this area we should probably remove: 
> https://github.com/node-usb/node-usb/blob/master/tsc/webusb/webusb-device.ts#L15

Thanks for the hint. I will extract this function and try it directly trying to catch the error.

amochkin, 2022-09-22:
@thegecko 
Ok, I figured it out.
1. The error was `LIBUSB_ERROR_ACCESS`.
2. I just completely forgot to add specific udev rule, duh.
3. Rule added, restarted, and voila - it works!

closed 2022-09-22 by amochkin

thegecko, 2022-09-22:
Awesome!

We shouldn't mask that error, I'll create a PR to remove the try...catch

amochkin, 2022-09-22:
Great, that will help the community.
Also I found a workaround:
before doing `await WebUSBDevice.createInstance(legacyDevice);`
one could try doing `legacyDevice.open();`
This will throw if something goes wrong.