node-usb/node-usb #546: Migrating from `usb-detection`: How to get Serial Number

hrueger, 2022-09-22:
Hi,
I'm trying to migrate from `usb-detection`. I have lot of devices with the same VID and PID and I need to differenciate them by their serial number. Using `usb-detection` I could get the serialNumber directly in the callback:
```TS
usbDetect.on("add", (newDevice) => console.log(newDevice.serialNumber));`
```

If I read the docs correctly, I need to open the USB Device in order to read its serial number, right? This feeld wrong, because there might be another process trying to open that device, too. I don't want to just claim every usb device with that VID and PID...
Did usb-detection open them in the background?

This is probably similar to #328, but I can't reopen that one.

thegecko, 2022-09-22:
> I need to open the USB Device in order to read its serial number, right?

Yes, the serial number string is not available in the usb descriptors, so you need to read it.

Other things which may help:

- turn it into a WebUSB device first which will populate this (may be too heavy)
- use (or look at) the [findBySerialNumber function](https://github.com/node-usb/node-usb/blob/master/tsc/index.ts#L24)

hrueger, 2022-09-22:
Wow, thanks for that quick answer 🎉
> turn it into a WebUSB device first which will populate this (may be too heavy)
does this internally open the device?

I wonder if `usb-detection` also opened the device... Maybe @Julusian knows that? (sorry for pinging you ;-) )

> use (or look at) the [findBySerialNumber function](https://github.com/node-usb/node-usb/blob/master/tsc/index.ts?rgh-link-date=2022-09-22T08%3A17%3A20Z#L24)
thanks, I see that it also opens the device. I'll try that.

hrueger, 2022-09-22:
I just did a quick test using the code of the [`findBySerialNumber` function](https://github.com/node-usb/node-usb/blob/master/tsc/index.ts?rgh-link-date=2022-09-22T08%3A20%3A30Z#L24).
However, I get
```
Error: LIBUSB_ERROR_NOT_SUPPORTED
    at Device.ExtendedDevice.open (A:\_Source\GLUE\node_modules\usb\tsc\usb\device.ts:68:14)
    at Discoverer.getSerialNumber (A:\_Source\GLUE\packages\makehaus-lib\src\control\Discovery.ts:108:24)
    at EventEmitter.<anonymous> (A:\_Source\GLUE\packages\makehaus-lib\src\control\Discovery.ts:64:39)
    at EventEmitter.emit (node:events:390:28)
    at EventEmitter.emitHotplugEvents (A:\_Source\GLUE\node_modules\usb\tsc\usb\index.ts:63:17)
    at EventEmitter.emit (node:events:390:28) {
  errno: -12
}
```

If I read #182 correctly, that means that there's a wrong driver. Unfortunately, I can't require users to use Zadig to change the driver, just that I can read the serial number...
I only need to find the serial number of the device. The physical usb device provides a virtual com port which is used by [`serialport`](https://github.com/serialport/node-serialport).

thegecko, 2022-09-22:
> does this internally open the device?

Yes

> I wonder if usb-detection also opened the device... 

It will (or Windows will when you plug it in)

This library uses libusb to be compatible across operating systems. Windows requires a driver for usb devices and can load winusb drivers for common devices (including HID devices AFAIK). 

On windows you can also try:

https://github.com/node-usb/node-usb#usbuseusbdkbackend

Julusian, 2022-09-22:
It looks like `usb-detection` is using os native methods to fetch the serialnumber for each device. 

Parsing the unique device id on windows: https://github.com/MadLittleMods/node-usb-detection/blob/master/src/detection_win.cpp#L442-L467
Linux exposes it as part of a set of properties https://github.com/MadLittleMods/node-usb-detection/blob/master/src/detection_linux.cpp#L153-L155
Mac has an api to call for it https://github.com/MadLittleMods/node-usb-detection/blob/master/src/detection_mac.cpp#L189

So it appears that it is possible on some platforms to retrieve the serialnumber without opening the device. But perhaps libusb isnt exposing a way to access this information?

Julusian, 2022-09-22:
Looking upstream, it sounds like there is no technical reason it isnt possible, and there has been some interest in supporting it, but nobody has provided a complete implementation yet. 

https://github.com/libusb/libusb/issues/866
https://github.com/libusb/libusb/pull/875

hrueger, 2022-09-22:
Thanks for your hints. I'm gonna stay with @Julusian's patch of the usb-detection lib for now.
Maybe we can leave this open until some implements it upstream?

mildsunrise, 2022-09-22:
hi, just wanted to say

> This feeld wrong, because there might be another process trying to open that device, too

there's nothing wrong with multiple processes accessing a device at the same time. as long as you don't do stuff such as resetting the device, changing to another configuration, or claiming an interface, you shouldn't interfere with other processes.

your use case in particular (opening a set of usb devices to get their serial number) is something I've definitely seen done in other libraries

hrueger, 2022-09-23:
Interesting. I thought that (at least on Windows) USB devices would be opened in exclusive mode (or was that only with hid devices? I can't remember.)
Even if that did not cause trouble, I'm still stuck with the LIBUSB_NOT_SUPPORTED error.

thegecko, 2022-10-31:
I would be happy to expose/utilise functionality to get the usb serial number on windows if someone were to port it from `usb-detection`.

e.g., the `findBySerialNumber` function could use native Windows code on windows and libusb otherwise. This feels like it would really harden the library and avoid a lot of libusb crashes finding devices on Windows. @Julusian  WDYT?

hrueger, 2022-11-01:
> on windows 

What would speak agains exposing that functionality on MacOS and Linux, too?

thegecko, 2022-11-01:
> What would speak agains exposing that functionality on MacOS and Linux, too?

I believe on those platforms you would be recreating how libusb does it.

NoahAndrews, 2023-03-22:
> > What would speak agains exposing that functionality on MacOS and Linux, too?
> 
> I believe on those platforms you would be recreating how libusb does it.

Does this mean that on macOS and Linux you can retrieve the serial number without calling `open()`?

Julusian, 2023-03-22:
For those watching, there is movement on this upstream https://github.com/libusb/libusb/pull/1258.
I am keeping an eye on it in case I can help either there or here

> I would be happy to expose/utilise functionality to get the usb serial number on windows if someone were to port it from usb-detection.
>
> e.g., the findBySerialNumber function could use native [...]

I would treat anything from node-usb-detection as questionable, as there are probably newer apis that will make the task simpler (as was the case for hotplug detection), but it could be done if someone is motivated to do it.

NoahAndrews, 2024-01-23:
The libusb PR got closed and its branch deleted :(

qwertychouskie, 2024-10-01:
https://github.com/libusb/libusb/pull/1532 is the current version being worked on.

ItsHarper, 2025-08-28:
Just shy of 3 years, the day has finally come: https://github.com/libusb/libusb/commit/e894c3e17a00c489934b8fc23edb9cce939d8503

I no longer even have much reason to care haha