node-usb/node-usb #547: Windows 7 support?

chsing, 2022-09-22:
Not so much a bug report as a question for clarification: does this package support Windows 7? I saw in [this PR](https://github.com/node-usb/node-usb/pull/524) comments about it failing on Windows 7, but can't find any documentation anywhere that confirms this, so just wanted to get a confirmation.

Asking because, I'm getting this error when trying to run my Electron app on Windows 7:
[image]

It runs fine on Windows 10.

Thanks!

## Additional Information
- Operating System: Windows 7
- Runtime (`node -v` or `electron -v`): node v16.14.2, Electron v19.0.0
- Node USB Version (`npm ls usb`): v2.5.1

thegecko, 2022-09-22:
Hmm,

There's a prebuilt binary for win32 created here:

https://github.com/node-usb/node-usb/blob/master/.github/workflows/prebuild.yml#L23

It uses the oldest windows runner available, but I don't have access to any Windows 7 machines to test. It should work in theory, but your issue suggests it doesn't.

Feel free to determine the issue otherwise I'm not sure we will get any traction from others to fix this :/

chsing, 2022-09-22:
Okay, thanks for the info!

closed 2022-09-22 by chsing