node-usb/node-usb #548: When using `startPoll` with multiple transfers, and error event can cascade mutliple times

strazzere, 2022-09-22:
## Bug Description:

## Steps to Reproduce:
1. Connect a usb device, perform a `startPoll` with multiple transfers; `epIn.startPoll(6)`
2. Unplug the device, this will cause a `LIBUSB_TRANSFER_ERROR` which will drop into this call https://github.com/node-usb/node-usb/blob/master/tsc/usb/endpoint.ts#L106
3. Multiple `Polling is not active.` are thrown, from different threads and are seemingly not all catchable. 

I'm unsure if this is expect behavior exactly. I would propose that we can wrap the `this.stopPoll()` call with a check to see if polling is still active, e.g. -

```
            if (!error) {
                this.emit('data', buffer.slice(0, actualLength));
            } else if (error.errno != LIBUSB_TRANSFER_CANCELLED) {
                this.emit('error', error);
                if (this.pollingActive) {
                    this.stopPoll();
                }
            }
```

This would allow a user to catch and handle all the errors while they are being emitted, and allow us to internally clean up the transfer once, which is all that is seemingly needed.

Let me know what you think as I can draft a PR for this if you agree on the solution. I may be misunderstanding the lifecycle here, but it /feels/ like it is an issue that should be handled that way.

thegecko, 2022-10-31:
Thanks for this bug report and apologies in the delay getting back to you.

I don't think this is intended behaviour and may have been introduced with https://github.com/node-usb/node-usb/pull/522

Your proposed solution looks sensible, would you mind trying that and opening a PR if it fixes the issue?

strazzere, 2022-10-31:
PR Submitted, it is currently waiting approval before the ci/cd will pick it up.

commit 6e049885 (2022-10-31): Fix #548 allow multiple polling to fail. (#556)

Allow multiple transfers to emit errors,
however only attempt to stop polling if
the polling is currently active.
commit 9d6dcdad (2023-08-22): Fix startPolling to use callback vs over emitted errors. (#631)

* Fix #548 allow multiple polling to fail.

Allow multiple transfers to emit errors,
however only attempt to stop polling if
the polling is currently active.

* Ensure callback is utilized, prevent FATAL_ERROR

The way this originally worked, it was possible to
over emit errors, which can cause node to lose track
of who can catch these. This specifically can cause many
`FATAL ERROR:  LIBUSB_TRANSFER_ERROR` or sometimes stalls.

* Fix linter issue

* Add test to ensure callback is occuring

* Move done assertion in test to call ba [... cut]
closed 2022-10-31 by thegecko