node-usb/node-usb #552: Detach event does not fire if device was connected when module was required

Hexxeh, 2022-10-06:
## Bug Description:

The first detach event doesn't fire if the device was already connected when the process started. This doesn't reproduce if I unplug the device before starting the process and then plug it in afterwards.

## Steps to Reproduce:
1. Plug in a USB device
2. Start a Node process that adds detach/attach listeners
3. Unplug the device, detach event should fire but does not

## Additional Information
- Operating System: macOS Monterey
- Runtime (`node -v` or `electron -v`): Node 16.16
- Node USB Version (`npm ls usb`): 2.5.2 (also tried 2.4.3)

thegecko, 2022-10-07:
I've been unable to reproduce this with the latest release, both with devices attached before and after running node on macOS.

I used this, do you have a similar reproducer?

```javascript
const usb = require('usb').usb;

usb.on('attach', device => console.log('attach', device));
usb.on('detach', device => console.log('detach', device));

setTimeout(process.exit, 10000);

```

Hexxeh, 2022-10-07:
Ah, sorry, I forgot to mention I'm using the WebUSB interface not the legacy one:

```
const WebUSB = require('usb').WebUSB;

const webUSB = new WebUSB({ allowAllDevices: true });

webUSB.addEventListener('connect', device => console.log('connect', device));
webUSB.addEventListener('disconnect', device => console.log('disconnect', device));
```

I tried the snippet you provided and that works as I'd expect, so this seems to be a WebUSB interface issue specifically.

Hexxeh, 2022-10-07:
Though even with the `usb.on` event, it seems not to fire if the device is open and actively being used (open/selectConfiguration/claimInterface) when disconnected. My use case is that I'm trying to cleanup references to the device and mark it disconnected when the user unplugs it.

thegecko, 2022-10-07:
Due to the webusb device requiring more data to be held than is available once the device is disconnected, the backing "store" of devices for it will need to be initiliased as your app starts.

I don't want to do this automatically as this would bring in unnecessary device listing of all potential devices.

Even doing it as the webusb clas is constructed would introduce an async constructor :(

Try listing the websub devices first which will initialise the list of known devices:

```javascript
const WebUSB = require('./').WebUSB;

const webUSB = new WebUSB({ allowAllDevices: true });

(async () => {
    await webUSB.getDevices();
    webUSB.addEventListener('connect', device => console.log('connect', device));
    webUSB.addEventListener('disconnect', device => console.log('disconnect', device));
    
    setTimeout(process.exit, 10000);    
})();
```

This may need to be documented or exposed as an 'initialise' call for webusb...

closed 2023-02-26 by thegecko