node-usb/node-usb #554: Does not see updated product id from within a Flatpak

jwillikers, 2022-10-23:
## Bug Description:

I'm working on a Flatpak of a keyboard utility, [Chrysalis](https://github.com/keyboardio/Chrysalis), which uses `node-usb`.
As part of the firmware upgrade process, the keyboard is flipped into bootloader mode by the application.
Once this occurs, the keyboard reconnects with an updated product id.
At this point, the application continues to query the USB devices with `getDeviceList()` until it locates the product id indicating the bootloader.
However, `getDeviceList()` continues to report the old product id instead of the new one.
I confirmed from inside the Flatpak the Flatpak that the updated product id was shown in the corresponding `/sys/bus/usb/devices/*/idProduct` file and that the old product id was no longer available there.

## Steps to Reproduce:
1. Use `getDeviceList()` to retrieve a list of the available USB devices from inside a Flatpak.
2. Flip a device into bootloader mode, causing the product id.
3. Use `getDeviceList()` to retrieve an updated list of the USB devices.

I can provide more details to reproduce if need be.

## Additional Information
- Operating System: Linux Flatpak -> Freedesktop Platform 22.08
- Runtime (`node -v` or `electron -v`): Node 16 Electron 17.4.6
- Node USB Version (`npm ls usb`): 2.5.1

The PR is keyboardio/Chrysalis#708.
The manifest used to build the Flatpak can be found [here](https://github.com/jwillikers/flathub/tree/io.keyboard.Chrysalis).
Flatpaks contain only a limited `libudev` in their sandbox for security purposes.
The `libudev` library within the Freedesktop Platform runtime may not be reliable given udev doesn't promise ABI stability and the version may not match that used on the system itself.

I've provided my own version of `libusb` within the Flatpak, which I tell `node-usb` to use by adding `use_system_libusb=true` to `GYP_DEFINES`.
I also disable udev for `node-usb` by including `use_udev=0` in `GYP_DEFINES`.

thegecko, 2022-10-31:
I'm not aware we cache the list of devices anywhere and always request this directly from libusb:

https://github.com/node-usb/node-usb/blob/master/src/node_usb.cc#L104

This may be an issue in libusb?

mildsunrise, 2022-11-01:
in the USB 2.0 standard, there's nothing explicitly allowing nor prohibiting the device from returning a different device description once setup...

and libusb definitely assumes the descriptor will never change, as (at least in the Linux backend) it's only read from the kernel in [`initialize_device`](https://github.com/libusb/libusb/blob/c060e9ce30ac2e3ffb49d94209c4dae77b6642f7/libusb/os/linux_usbfs.c#L901), which is called from `linux_enumerate_device` (and only when the device doesn't already exist in the context) and `op_wrap_sys_device`. there's apparently no way to cause the descriptor to be re-read other than closing + unreffing the device and listing it again.

so clearly, if what you say is true - if this USB device is returning different device descriptors once in configured state - then the device is doing something that... may not be illegal per se, but is at least completely unsupported by the ecosystem. are you sure that the `*` part in `/sys/bus/usb/devices/*/idProduct` does NOT change when the device switches to the new productId?

jwillikers, 2022-11-01:
@mildsunrise I left the `*` in there to just to indicate the general path. I grepped through the usb devices tree without concern for the actual id where the `*` was. I'd think that the device would have completely disappeared and been re-configured when switching to bootloader mode, but I will check to see whether the id under `/sys/bus/usb/devices/` changes or stays the same and let you know.

jwillikers, 2022-11-01:
I can confirm that the USB device is located at `/sys/bus/usb/devices/1-5.3.3/`. When switched into bootloader mode, the path remains the same, though the product id changes. If I unplug the keyboard and plug it back in, it shows up at the same path.

mildsunrise, 2022-11-02:
ah, my bad. I forgot that the `*` in sysfs is the topological path, not the address,
could you re-test, but this time looking at `/sys/bus/usb/devices/*/devnum` to see if it changes or stays the same?

mildsunrise, 2022-11-02:
also, in each of both cases, could you call `getDeviceList()` before and after, and check the `.deviceAddress` of the returned devices?

jwillikers, 2022-11-03:
Before:

```
$ cat /sys/bus/usb/devices/1-4.3.3/devnum
15
```

```
Vendor ID: 13462, Product ID: 6, Device Address: 15
```

After:

```
$ cat /sys/bus/usb/devices/1-4.3.3/devnum
25
```

My logging was kind of ad-hoc, which is why this number is not just `16`.
I didn't press the necessary key to flip into bootloader mode in order to log the values for the device address before flipping it into bootloader mode. It still resets the keyboard which then comes back as a new `devnum` each time. It started out as `15` and was `24` right before I actually flipped it into bootloader mode, so it received a new device number. If I just flip it into bootloader mode the first time, it shows up as a new device number just the same. Either way, the output below will continue to show the original device address an product ID.

```
Vendor ID: 13462, Product ID: 6, Device Address: 15
```

mildsunrise, 2022-11-03:
and when you reset the keyboard, does the list of devices returned by node-usb change? if you look at the `deviceAddress` property (which should be equivalent to the `devnum` file), is node-usb reporting the old device or the new one?

jwillikers, 2022-11-03:
`node-usb` reports the old device `devnum` number after resetting the keyboard. The `Vendor ID: 13462, Product ID: 6, Device Address: 15` is the logged output from `node-usb`.

mildsunrise, 2022-11-03:
interesting, I wonder what happens if libusb is built without udev support...

mildsunrise, 2023-02-27:
I don't think there's anything I can do here... I remember doing some simple-ish tests to try and reproduce the issue, but no luck. if there's a way to reproduce that doesn't need specific hardware (I also have an STM32F4 if it helps) let me know.