node-usb/node-usb #555: Error: The specified procedure could not be found. (Windows7)

tombertrand, 2022-10-25:
[image]

## Bug Description:

We were reported this error on Windows7 after updating to `Electron 19.1.2` and `Node USB 2.5.0`.

Unfortunately I don't have any specifics other than these. Opening this ticket in case others search for it or a maintainer has an idea of what could be the cause.

## Additional Information
- Operating System: Windows7
- Electron 19.1.2
- Node USB 2.5.0

thegecko, 2022-10-31:
Erk, we don't really support Windows 7 (not sure about libusb either), primarily because I don't have access to that OS for testing.

Would be useful to know if it was the electron upgrade or node-usb upgrade which caused the issue.

It should be easy to create a test app to try node-usb on the client's machine to help pinpoint the program (and version) with the issue.

baversjo, 2023-01-19:
@thegecko @tombertrand  I also had this issue. It occurs when attempting to load this lib. When I remove the require('usb') from my electron code, my app works.

baversjo, 2023-01-19:
It says here that libusb should support windows 7: https://github.com/libusb/libusb/wiki/Windows#usb-3x-support

Windows 7 support is not a priority for me but would be nice to figure this out!

Julusian, 2023-01-24:
I believe I broke that https://github.com/node-usb/node-usb/pull/524#issuecomment-1188924704

Maybe it would be possible to try to load the required windows api symbols and gracefully fail. 
But I personally don't think it is worth the effort for a platform which was EOL 2 years ago, is not supported by any current version of nodejs and electron has [announced ending support](https://www.electronjs.org/blog/windows-7-to-8-1-deprecation-notice)

baversjo, 2023-01-24:
I think you’re right @Julusian , I wanted to point out that I could solve this issue for my users still on win 7 by using usb 1 instead of usb 2. I don’t like having usb 1 in my package.json but users are happy :)

Julusian, 2023-01-24:
@baversjo you should be able to use any version before v2.5.0
v2.4.3 does not include my changes, so I expect will work on windows 7

thegecko, 2023-01-24:
I think we've agreed to not support Windows 7. @Julusian  feel free to add support if it bugs you in future ;)

I'll close this issue with the known workaround to use v2.4.3 or earlier.

closed 2023-01-24 by thegecko