node-usb/node-usb #559: electron-builder fail

ifzm, 2022-11-17:
## Bug Description:
```shell
> electron-builder

  • electron-builder  version=23.6.0 os=10.0.22000
  • loaded configuration  file=D:\Code\Workspace\scc\scc-video-upload-client\electron-builder.json5
  • writing effective config  file=release\0.0.1\builder-effective-config.yaml
  • rebuilding native dependencies  dependencies=usb@2.5.2 platform=win32 arch=x64
  ⨯ cannot execute  cause=fork/exec C:\Users\Jim\AppData\Roaming\nvm\v16.14.2\node_modules\pnpm\bin\pnpm.cjs: %1 is not a valid Win32 application.
                    command='C:\Users\Jim\AppData\Roaming\nvm\v16.14.2\node_modules\pnpm\bin\pnpm.cjs' rebuild usb@2.5.2
                    workingDir=
 ELIFECYCLE  Command failed with exit code 1.
```

## Steps to Reproduce:
1. run success
2. build fail

## Additional Information
- Operating System: Windows 11
- Runtime (`node -v` or `electron -v`): 21.3.0
- Node USB Version (`npm ls usb`):2.5.2

thegecko, 2022-11-17:
All common native binaries are shipped with node-usb (supporting electron and native node) and you shouldn't be rebuilding them.

ifzm, 2022-11-23:
thanks :) I need to configure npmrebuild to false in electron-builder

closed 2022-11-23 by ifzm