node-usb/node-usb #570: isAlloweddevice only match when `productId`, `vendorId` and `serialNumber` are the same

adriencarbonaro, 2023-02-09:
The `USBDeviceFilter` indicates that properties are optional (`@types/w3c-web-usb/index.d.ts`) : 
```typescript
interface USBDeviceFilter {
    vendorId?: number | undefined;
    productId?: number | undefined;
    classCode?: number | undefined;
    subclassCode?: number | undefined;
    protocolCode?: number | undefined;
    serialNumber?: string | undefined;
}
```
So setting only one property should mean all devices with this property will be allowed.

But in the `isAllowedDevice` function, 3 properties are mandatory to match devices.

https://github.com/node-usb/node-usb/blob/556346d23830f0c00098b2f984d9283b5b5f332c/tsc/webusb/index.ts#L360-L362

So there is no way to just allow device with specific `vendorId`.

thegecko, 2023-02-11:
Hi @adriencarbonaro . Apologies the code is quite complicated, but there are two lots of filtering going on.

The first is the standard WebUSB `requestDevice()` filter using `USBDeviceFilter` which is checked here: 

https://github.com/node-usb/node-usb/blob/master/tsc/webusb/index.ts#L300

Next, in a node environment the browser context does not exist to remember previous authorised devices.

Therefore we have added the extra [allowedDevices](https://github.com/node-usb/node-usb/blob/master/tsc/webusb/index.ts#L17) option to allow the user to pre-authorise devices. This is not in the WebUSB spec. and is checked in the code you linked: 

https://github.com/node-usb/node-usb/blob/master/tsc/webusb/index.ts#L359

The scenario you outline which may fail actually exists in the tests, so I'm pretty confident it works as intended:

https://github.com/node-usb/node-usb/blob/master/test/webusb.coffee#L15

However, if you have a specific situation which isn't working, could you outline it here?

adriencarbonaro, 2023-02-13:
## OK

*index.ts*
```typescript
import * as usb from "usb";

const wusb = new usb.WebUSB({
	allowAllDevices: true
});

wusb.getDevices().then(device => { console.log(device); });
```

*output*

```
[
  WebUSBDevice {
    device: {
      busNumber: 1,
      deviceAddress: 2,
      deviceDescriptor: [Object],
      portNumbers: [Array],
      interfaces: undefined
    },
    configurations: [ [Object] ],
    usbVersionMajor: 2,
    usbVersionMinor: 0,
    usbVersionSubminor: 0,
    deviceClass: 239,
    deviceSubclass: 2,
    deviceProtocol: 1,
    vendorId: MY_VID,
    productId: MY_PID,
    deviceVersionMajor: 2,
    deviceVersionMinor: 0,
    deviceVersionSubminor: 1,
    manufacturerName: 'manufacturerName',
    productName: 'ProductName',
    serialNumber: 'sn'
  }
]
```

## NOK

*index.ts*
```typescript
import * as usb from "usb";

const wusb = new usb.WebUSB({
	allowedDevices: [
		{ vendorId: MY_VID }
	]
});

wusb.getDevices().then(d => {
	console.log(d);
})
```
*output*
```
[]
```

thegecko, 2023-02-13:
OK, so you want to be able to just pass a single item when using the allowedDevices? Perhaps I was wrong to reuse the `USBDeviceFilter` types.

I'll create a PR and see if this fits your needs. This won't change the standard `requestDevice` functionality as outlined in the specification, but will affect the allowedDevices extra option.

adriencarbonaro, 2023-02-13:
I see 2 goals when using the lib : 
1. Allow only a specific device, for which you know all the properties. This is working fine right now.
2. Allow all "your" devices (i.e. with the same VID). In this case you might not know all the properties of the filter (*ex:* the serial number).

*My case specifically:*  I'm writting an app that allows to upgrade all my devices (same VID/PID). I don't know their serial number in advance, as it is randomly generated on production.

thegecko, 2023-02-21:
@adriencarbonaro could you take a look at https://github.com/node-usb/node-usb/pull/574 ?
I added a test which I believe replicates your scenario

closed 2023-02-26 by thegecko

adriencarbonaro, 2023-02-27:
@thegecko it works, thank you ! 👍

PR #574 (2023-02-26): Rework options.allowedDevices
## Changes

Reworked filtering to correctly use the `allowedDevice` filters

## Fixes

Fixes #570

## Checklist

Have you...

- [x] Tested the change acts as expected
- [x] Checked the change doesn't remove or change existing functionality
- [x] Considered how the feature impacts the product and tested around it (not just the happy paths)
- [x] Where possible, executed the hardware tests on multiple operating systems