node-usb/node-usb #571: Missing TS declarations in npm package

hrueger, 2023-02-10:
## Bug Description:
VSCode can't detect the `index.d.ts` file because it is not referenced in the `package.json`'s types` field.

## Steps to Reproduce:
1. Use VSCode
2. Install `usb` as a dependency in a TS project
3. See an error.

## Additional Information
- Operating System: n/a
- Runtime n/a
- Node USB Version (`npm ls usb`): latest

thegecko, 2023-02-10:
The `types` package entry is only required if the types aren't alongside the output javascript files. In this package they are.

I use this package in VS Code and have intellisense, etc. fine. What error do you see?

hrueger, 2023-02-10:
Maybe it only happens if the package is hoisted? I'm using yarn workspaces.

Edit: never mind, I can't reproduce it anymore... Very strange. Probably VSCode tried to detect it while it was still installing and therefore could not find the `.d.ts` file.

closed 2023-02-10 by hrueger