node-usb/node-usb #573: [FR] WebUSB requestDevice should throw special error for no device found + small doc issue

yume-chan, 2023-02-13:
### Feature Description:

I made a ADB (Android Debugging Bridge) [implementation](https://github.com/yume-chan/ya-webadb) for browsers using WebUSB API. Now I want to make that compatible with this package for using in Node.js environment.

When no device found from `requestDevice`, browser throws an `DOMException` with `name: "NotFoundError"`, so I now use `e instanceof DOMException && e.name === "NotFoundError"` to detect such error. But in this package, it only throws a normal `Error`. 

https://github.com/node-usb/node-usb/blob/f3ced8398d4214ba1c7d05ca129aba1d5f6e21f8/tsc/webusb/index.ts#L194-L196

I know there aren't many other `Error`s would be thrown from here, but I still hope I can handle them precisely. Node.js has `DOMException` on `globalThis` since version 17 (https://nodejs.org/api/globals.html#domexception). [dom-exception](https://www.npmjs.com/package/node-domexception) NPM package can retrieve the hidden `DOMException` class since Node.js 10.0.5.  A normal `Error` with a `name: "NotFoundError"` field would also be fine.

### Plus a doc issue

README says:

https://github.com/node-usb/node-usb/blob/f3ced8398d4214ba1c7d05ca129aba1d5f6e21f8/README.md?plain=1#L135-L141

`device` should never be falsy, an Error will be thrown in this case.

mildsunrise, 2023-02-13:
yes, this is one of the things in my todolist. this package should ideally throw DOMException with `name` set as per the spec.
as you indicated, we'd need some kind of polyfill to get access to DOMException for Node.js < 17.
I don't think I'll have time to implement this myself, though :pensive:

yume-chan, 2023-02-15:
node-domexception package doesn't support Node.js 10.x https://github.com/jimmywarting/node-domexception/issues/1

How about copy-pasting the source code into this package?

thegecko, 2023-02-21:
Thanks for your feedback @yume-chan 

Please see https://github.com/node-usb/node-usb/pull/575

`DomException` is declared in `lib.dom.ts` so seems to just work?

yume-chan, 2023-02-22:
> DomException is declared in lib.dom.ts so seems to just work?

A Node.js package shouldn't use `lib.dom.d.ts`. A type exists in type definition doesn't mean it exists at runtime. `DOMException` was exposed in Node.js version 17 so it doesn't work in older versions.

yume-chan, 2023-02-26:
> I've updated the PR to stick with standard Errors for now, but added the correct name for checking.

It's OK for me.

thegecko, 2023-02-26:
Ok, I've merged it. I'll close this issue for now, we can reopen it if we need to consider implementing the DomException.

closed 2023-02-26 by thegecko