node-usb/node-usb #597: Error when listening to Usb connect events in electron

akashdeep-sharma, 2023-04-18:
## Bug Description:
Getting this error 
`Uncaught Exception:
Error: No native build was found for platform=darwin arch=arm64 runtime=electron abi=114 uv=1 armv=8 libc=glibc node=18.14.0 electron=24.1.2 webpack=true
 `
 

## Steps to Reproduce:
1. using WebUsb with allowedAllDevices set to true
2.Listening to 'connect ' and 'disconnect' events 
3.getting this error after building electron app and running it. App works fine when I don't listen to events from usb package 

`  const webUsb = new usb.WebUSB({
      allowAllDevices: true,
    });
    webUsb.addEventListener('connect', (event) => {
      console.log(event);
    });`
code snippet

thegecko, 2023-04-18:
There should be a valid prebuild for your environment, are you trying to rebuild the package in any way?

akashdeep-sharma, 2023-04-18:
Yes I tried rebuilding
this is my package command
`    "package": "ts-node ./.erb/scripts/clean.js dist && npm run build && electron-builder build --publish never && electron-rebuild -w usb",
`

thegecko, 2023-04-18:
Yes, don't rebuild the package. The correct prebuild should already be shipped. Rebuilding it causes issues

akashdeep-sharma, 2023-04-19:
Without rebuilding also I am getting same error

thegecko, 2023-04-19:
OK, can you confirm you see the same issue with our electron test-harness:

https://github.com/node-usb/node-usb-example-electron