node-usb/node-usb #603: switch out eventemitter for eventtarget

jimmywarting, 2023-04-29:
you know... in order to follow the spec more...

mildsunrise, 2023-04-29:
dropping support for Node.js < 14.5.x just for a detail in the hotplug detection API seems not worth it for me... i'd give more priority to stuff like switching Error for DOMError imho

jimmywarting, 2023-04-29:
node v20 just got released so support for v14 will end in 2023-04-30
v16 is the LTS now.

mildsunrise, 2023-04-29:
yeah but this is a big project and y'know. not sure what we're really supporting in practice, but at least in theory we're adamant to drop support for old Nodes. not saying I wouldn't consider it though. @thegecko thoughts?

thegecko, 2023-05-01:
I agree @mildsunrise. While I'm not adamant we support older node versions forever, I'm also keen we break people as little as possible.

Are there any other benefits we can get through dropping older node support which would make this worthwhile?

thegecko, 2023-05-01:
I'm wondering if we can at least drop node 10 support and allow to N-API 8 features