node-usb/node-usb #625: caught Error: Dynamic require of "util" is not supported

yoyos, 2023-05-31:
## Bug Description:
Can't use node-usb sample code in an electron + react app + win10 + nodeJS 18.15

```
const devices = getDeviceList();
for (const device of devices) {
    console.log(device); // Legacy device
}
```

```
chunk-OROXOI2D.js?v=0b4b26d5:10 Uncaught Error: Dynamic require of "util" is not supported
    at chunk-OROXOI2D.js?v=0b4b26d5:10:9
    at node_modules/.vite-electron-renderer/util.mjs (util.mjs:1:50)
    at __init (chunk-OROXOI2D.js?v=0b4b26d5:13:56)
    at node_modules/usb/dist/index.js (index.ts:1:1)
    at __require2 (chunk-OROXOI2D.js?v=0b4b26d5:16:50)
    at index.ts:74:1
```

## Steps to Reproduce:
1. npm create electron-vite , select React
2. npm i
3. npm i usb
4. add sample code to App.tsx with correct import
```
const devices = getDeviceList();
for (const device of devices) {
 console.log(device);
}
```
5. npm run dev

## Additional Information
- Operating System: Windows 10
- Runtime (`node -v` or `electron -v`): NODEJS v18.15.0
- Node USB Version (`npm ls usb`): `-- usb@2.9.0
- Other dependencies:

```
"dependencies": {
    "react": "^18.2.0",
    "react-dom": "^18.2.0",
    "usb": "^2.9.0"
  },
  "devDependencies": {
    "@types/react": "^18.0.37",
    "@types/react-dom": "^18.0.11",
    "@typescript-eslint/eslint-plugin": "^5.59.0",
    "@typescript-eslint/parser": "^5.59.0",
    "@vitejs/plugin-react": "^4.0.0",
    "electron": "^24.4.0",
    "electron-builder": "^23.6.0",
    "eslint": "^8.38.0",
    "eslint-plugin-react-hooks": "^4.6.0",
    "eslint-plugin-react-refresh": "^0.3.4",
    "typescript": "^5.0.2",
    "vite": "^4.3.2",
    "vite-plugin-electron": "^0.11.2",
    "vite-plugin-electron-renderer": "^0.14.5"
  },
```

yoyos, 2023-05-31:
I think I was putting the code in a wrong place in electron. 
If I put the code in the main.js of electron, another error appear:

```
App threw an error during load
Error: No native build was found for platform=win32 arch=x64 runtime=electron abi=114 uv=1 libc=glibc node=18.14.0 electron=24.4.0
```

thegecko, 2023-06-01:
We use an example application to test this library in electron over at:

https://github.com/node-usb/node-usb-example-electron

I've just updated this to use the latest `node-usb` and `electron` versions.

Please try this as a starting point and ensure you aren't rebuilding the native usb library at all.

closed 2023-06-06 by yoyos