node-usb/node-usb #629: LIBUSB_ERROR_ACCESS when trying to open (hid) device

Mola19, 2023-06-07:
## Bug Description:
`LIBUSB_ERROR_ACCESS` is thrown when i try to open one of my devices (specifically a mousepad). The device work with hidapi and node-hid without problems and all other hid devices on my machine are fine.
I know that i can resolve this by giving admin rights, but i'd rather avoid that.

I don't think that the mousepad has any interfaces the system would want to hide, like a keyboard or mouse has.

Note: There is a good chance that this is an issue on my side

## Steps to Reproduce:
```js
const usb = require("usb")

let dev = usb.getDeviceList().filter(device => device.deviceDescriptor.idProduct == 0x1890 && device.deviceDescriptor.idVendor == 0x0B05)[0]
dev.open()
```

This then throws `LIBUSB_ERROR_ACCESS` 

## Additional Information
Error happend on two windows machines
- Operating System: Windows 10
- Runtime: Node 19.6.0/Node 20.2.0
- Node USB Version: 2.9.0

Also there is an issue with detecting some hid devices (mostly keyboard) in 1.9.2 (it throws `LIBUSB_ERROR_NO_DEVICE`).
I know this version is quite old, but based on npm it is still the most downloaded version by far.
I don't know if there a still plans to maintain it, i just thought i'd mention it.

thegecko, 2023-06-07:
I believe HID devices are explicitly claimed by the OS. For interacting with these kind of device, please look at the [node-hid](https://github.com/node-hid/node-hid) project.

Mola19, 2023-06-07:
@thegecko No, i have several other hid devices attached that work just fine. Just this one

Mola19, 2023-06-07:
The issue might be that this device has only one interface

thegecko, 2023-06-07:
Ok, sorry I was too hasty on closing this. We often get asked to support HID devices and are limited by what libusb can do.

> ...there is an issue with detecting some hid devices (mostly keyboard) in 1.9.2 ... I don't know if there a still plans to maintain it

We don't plan to maintain any older versions of node-usb, so recommend you update if possible. Is there something specific about 1.9.2 which keeps you locked to that version?

Off the top of my head, there's some things to try regarding Windows drivers mentioned in the [README](https://github.com/node-usb/node-usb#windows).

Not sure if you need to claim the interface first?

The native library mainly wraps libusb, so you may find more information over there:

- https://github.com/libusb/libusb/wiki/Windows
- https://github.com/libusb/libusb/wiki/FAQ

> The issue might be that this device has only one interface

Could be, although I've not seen this before

mildsunrise, 2023-06-08:
I can't help much because my Windows knowledge is limited, but to expand more on what @thegecko said: HID isn't a property of the device, but of a particular interface. [compatible] HID interfaces will be claimed by the OS's HID driver (which on Linux is `usbhid`). on Linux you should in any case be able to open the device, just not claim those interfaces.

in any case, rather than claiming HID interfaces from the OS, it's always better to use the dedicated HID APIs of the OS (as pointed out).