node-usb/node-usb #635: Error: Expected buffer for OUT transfer (based on bmRequestType) when executing clearHalt.

asivery, 2023-06-14:
## Bug Description:
It seems like there's a problem with WebUSB API's clearHalt implementation. Executing clearHalt throws an error:

```
Expected buffer for OUT transfer (based on bmRequestType)
```

From what I understand, the error originates at:
https://github.com/node-usb/node-usb/blob/e9d33d81c495e33875b0dd85942af6335d85a27c/tsc/usb/device.ts#L137-L139

Which is called by:
https://github.com/node-usb/node-usb/blob/e9d33d81c495e33875b0dd85942af6335d85a27c/tsc/webusb/webusb-device.ts#L261

In this line, the value `0` is passed as `data_or_length`, but since it's an OUT request, `controlTransfer` expects a buffer and fails.

## Steps to Reproduce:
1. Execute clearHalt on a stalled endpoint of a WebUSB-API USB device.

## Additional Information
- Operating System: Ubuntu 22.04 
- Runtime (`node -v` or `electron -v`): Electron 13.6.9
- Node USB Version (`npm ls usb`): 2.4.3 (however it seems like it might still be present in the newest version)

thegecko, 2023-06-17:
Yeah, I can see the issue.

Some possible ways to resolve this and keep the code clean:

- create a new transfer object in `clearHalt` and leave the device controlTransfer as is - duplicates code
- remove the length or buffer size check in device controlTransfer - loses a guard
- extend the data_or_length union type to also accept `false` to skip the checks

Thoughts or other suggestions?

cc @mildsunrise

asivery, 2023-06-18:
@thegecko Why not create a new empty buffer in `clearHalt`, and pass that to `controlTransfer`?

thegecko, 2023-06-18:
> @thegecko Why not create a new empty buffer in clearHalt, and pass that to controlTransfer?

I think that would only help with OUT endpoints. AFAICT the problem also exists for IN endpoints, too.

asivery, 2023-06-18:
@thegecko Do you mean when using `clearHalt` on IN endpoints?
From what I understand, `clearHalt` is always an OUT control transfer. The actual endpoint is passed in `wIndex`. Feel free to correct me on that if I'm wrong please.

thegecko, 2023-06-18:
> From what I understand, clearHalt is always an OUT control transfer. The actual endpoint is passed in wIndex. Feel free to correct me on that if I'm wrong please.

In this case your suggestion should work!

thegecko, 2023-06-18:
How does #636 look?

asivery, 2023-06-18:
@thegecko I think that it might fail because of what I mentioned above.
It doesn't matter if the endpoint you're running `clearHalt` on is an IN or OUT endpoint, clearHalt itself still goes through as an OUT control transfer.
The bug I'm reporting originated in this code:

https://github.com/asivery/node-mass-storage/blob/master/src/usb-mass-storage.ts#L158-L162

It's running `clearHalt()` on an IN endpoint, but the error was still 'Expected buffer for OUT transfer (based on bmRequestType)'

thegecko, 2023-06-18:
Oh I see, I was assuming clearHalt gets called on the endpojnt being cleared, but it's actually on `LIBUSB_RECIPIENT_ENDPOINT`

PR updated accordingly.

asivery, 2023-06-18:
@thegecko Looks good!

PR #636 (2023-06-18): Fix clearHalt for OUT endpoints
## Changes

- Fix clearHalt for OUT endpoints

## Fixes

- Fixes #635

## Checklist

Have you...

- [ ] Tested the change acts as expected
- [x] Checked the change doesn't remove or change existing functionality
- [x] Considered how the feature impacts the product and tested around it (not just the happy paths)
- [x] Where possible, executed the hardware tests on multiple operating systems
closed 2023-06-18 by thegecko