node-usb/node-usb #653: Cannot install usb on Raspberry Pi Debian Bullseye

iKK001, 2023-07-12:
Trying to make `usb` run on a Raspberry Pi (Debian Bullseye - release 2023-05-03)

I installed:
- node version v18.16.1
- yarn v1.22.19

I also installed the two necessary libs as follows:
sudo apt install libudev-dev
sudo apt install build-essential

But then, when I do a `yarn add usb` then I get the following error message (see below).

Any idea what is wrong ?

```
yarn add usb
yarn add v1.22.19
[1/5] Validating package.json...
[2/5] Resolving packages...
[3/5] Fetching packages...
[4/5] Linking dependencies...
warning " > ts-node@10.9.1" has unmet peer dependency "@types/node@*".
[5/5] Building fresh packages...
[1/4] ⠈ electron
[2/4] ⠈ esbuild
[3/4] ⠈ @serialport/bindings-cpp
error /home/pi/iKK/electron-sensorbox_trial001/node_modules/usb: Command failed.
Exit code: 1
Command: node-gyp-build
Arguments:
Directory: /home/pi/iKK/electron-sensorbox_trial001/node_modules/usb
Output:
node:events:491
      throw er; // Unhandled 'error' event
      ^

Error: spawn node-gyp ENOENT
    at ChildProcess._handle.onexit (node:internal/child_process:283:19)
    at onErrorNT (node:internal/child_process:476:16)
    at process.processTicksAndRejections (node:internal/process/task_queues:82:21)
Emitted 'error' event on ChildProcess instance at:
    at ChildProcess._handle.onexit (node:internal/child_process:289:12)
    at onErrorNT (node:internal/child_process:476:16)
    at process.processTicksAndRejections (node:internal/process/task_queues:82:21) {
  errno: -2,
  code: 'ENOENT',
  syscall: 'spawn node-gyp',
  path: 'node-gyp',
  spawnargs: [ 'rebuild' ]
}
```

## Additional Information
- Operating System: 
- Runtime (`node -v` or `electron -v`):
- Node USB Version (`npm ls usb`):

iKK001, 2023-07-14:
hmm could it be that node v18 is too much for the `usb` ??

thegecko, 2023-07-14:
> Error: spawn node-gyp ENOENT

looks like node-gyp is missing

iKK001, 2023-07-14:
isn't node-gyp part of the `usb` lib dependencies already ?

I also added node-gyp manually (newest version) - but no help.

What in fact did help, is going back to node v16 (instead of v18) !!!

Seems that `usb` (or its dependencies) is not ready for node 18

thegecko, 2023-07-14:
That's odd, node should be backwards compatible

iKK001, 2023-07-16:
odd yes, but still not working on Raspberry Debian Bullseye ! Any ideas why ?

matthewcobb, 2024-01-22:
I was also having an issue building an AppImage for Electron and running it on a raspberry pi5. The image would build just fine, but fail to load on the Pi itself. Downgrading my mac to v16.x and then running the electron build scripts to build AppImages using v16 fixed my issue.

matthewcobb, 2024-01-23:
For some more context if it's useful, this was the error I was getting: `node_modules/usb/build/Release/usb_bindings.node: invalid ELF header`.

```
# Trying to open an AppImage via terminal on Raspberry Pi5

/tmp/.mount_defendVAteTG/resources/app.asar.unpacked/node_modules/usb/build/Release/usb_bindings.node: invalid ELF header
    at process.func [as dlopen] (node:electron/js2c/asar_bundle:2:2131)
    at Object.<anonymous> (node:internal/modules/cjs/loader:1356:18)
    at Object.func [as .node] (node:electron/js2c/asar_bundle:2:2358)
    at Module.load (node:internal/modules/cjs/loader:1126:32)
    at node:internal/modules/cjs/loader:967:12
    at Function._load (node:electron/js2c/asar_bundle:2:13327)
    at Module.require (node:internal/modules/cjs/loader:1150:19)
    at require (node:internal/modules/cjs/helpers:121:18)
    at load (/tmp/.mount_defendVAteTG/resources/app.asar/node_modules/node-gyp-build/node-gyp-build.js:22:10)
    at Object.<anonymous> (/tmp/.mount_defendVAteTG/resources/app.asar/node_modules/usb/dist/usb/bindings.js:8:38)```

Julusian, 2024-01-23:
@matthewcobb 

> For some more context if it's useful, this was the error I was getting: node_modules/usb/build/Release/usb_bindings.node: invalid ELF header.

That looks like when building the electron package this library was recompiled. Unless you are doing that on a pi, it is almost certainly being rebuilt for the wrong platform.

The fix is to tell your electron build tooling to skip recompiling this library