node-usb/node-usb #672: Uncatchable LIBUSB_TRANSFER_NO_DEVICE

cleoo, 2023-09-16:
## Bug Description:
We use node-usb to drive escpos thermal printers. It sometimes happens that certain models that are faulty or come with faulty firmware disconnect randomly from the usb bus.

When the disconnection occurs, the exception generated is impossible to catch and stops nodejs, which is not really acceptable.

## Error logged

> FATAL ERROR:  LIBUSB_TRANSFER_NO_DEVICE
 1: 0xb02960 node::Abort() [node /home/kiosk/[A]/index.js]
 2: 0xa18149 node::FatalError(char const*, char const*) [node /home/kiosk/[A]/index.js]
 3: 0xa18152  [node /home/kiosk/[A]/index.js]
 4: 0xaccedb napi_fatal_error [node /home/kiosk/[A]/index.js]
 5: 0x7f9da5e15046  [/home/kiosk/[B]/node_modules/usb/prebuilds/linux-x64/node.napi.glibc.node]
 6: 0x7f9da5e2440b handleCompletion(Transfer*) [/home/kiosk/[B]/node_modules/usb/prebuilds/linux-x64/node.napi.glibc.node]
 7: 0x7f9da5e154bd Napi::ThreadSafeFunction::CallJS(napi_env__*, napi_value__*, void*, void*) [/home/kiosk/[B]/node_modules/usb/prebuilds/linux-x64/node.napi.glibc.node]
 8: 0xaca7e0  [node /home/kiosk/[A]/index.js]
 9: 0x1547da6  [node /home/kiosk/[A]/index.js]
10: 0x155a4d4  [node /home/kiosk/[A]/index.js]
11: 0x15486f8 uv_run [node /home/kiosk/[A]/index.js]
12: 0xa3fa35 node::SpinEventLoop(node::Environment*) [node /home/kiosk/[A]/index.js]
13: 0xb42c56 node::NodeMainInstance::Run() [node /home/kiosk/[A]/index.js]
14: 0xac89bc node::Start(int, char**) [node /home/kiosk/[A]/index.js]
15: 0x7f9dc8227cd0  [/usr/lib/libc.so.6]
16: 0x7f9dc8227d8a __libc_start_main [/usr/lib/libc.so.6]
17: 0xa3d03c  [node /home/kiosk/[A]/index.js]
## Steps to Reproduce:
1. Possibly disconnect a working device during an info transfer

## 
## Additional Information
- Operating System: Archlinux
- Runtime (`node -v` or `electron -v`): v16.17.1
- Node USB Version (`npm ls usb`): usb@2.10.0

cleoo, 2024-02-10:
Good morning,

I realize that the defect is difficult to reproduce but it does exist. Could someone do a verification pass to check if all low-level exceptions are relayed on the nodeJS side?

It's not very acceptable for a library to cause the program to fail without being able to catch any exception.

Thanks in advance.

thegecko, 2024-02-10:
Have you tried the generic [uncaughtexception](https://nodejs.org/api/process.html#event-uncaughtexception)

cleoo, 2024-03-01:
Thanks for this answer and yes, process.on "unhandledRejection" and "uncaughtException" are well defined but the handlers are never called.

thegecko, 2024-03-03:
I did a little digging into this and found we should be explicitly turning on the exception handling for NAPI.
I've done this in https://github.com/node-usb/node-usb/pull/738, so it would be good to know if this has helped at all.

cleoo, 2024-03-06:
Hi,
Without certainty that the modifications have been integrated in v2.12.0, for my part I have no improvements. The nodejs process is still interrupted without an exception being thrown.

| FATAL ERROR:  LIBUSB_TRANSFER_NO_DEVICE
|  1: 0xb02960 node::Abort() [node /home/[...]/index.js]
|  2: 0xa18149 node::FatalError(char const*, char const*) [node /home/[...]/index.js]
|  3: 0xa18152  [node /home/[...]/index.js]
|  4: 0xaccedb napi_fatal_error [node /home/[...]/index.js]
|  5: 0x7fc0e5e13e36  [/home/[...]/node_modules/usb/prebuilds/linux-x64/usb.glibc.node]
|  6: 0x7fc0e5e2303b handleCompletion(Transfer*) [/home/[...]/node_modules/usb/prebuilds/linux-x64/usb.glibc.node]
|  7: 0x7fc0e5e14856 Napi::ThreadSafeFunction::CallJS(napi_env__*, napi_value__*, void*, void*) [/home/[...]/node_modules/usb/prebuilds/linux-x64/usb.glibc.node]
|  8: 0xaca7e0  [node /home/[...]/index.js]
|  9: 0x1547da6  [node /home/[...]/index.js]
| 10: 0x155a4d4  [node /home/[...]/index.js]
| 11: 0x15486f8 uv_run [node /home/[...]/index.js]
| 12: 0xa3fa35 node::SpinEventLoop(node::Environment*) [node /home/[...]/index.js]
| 13: 0xb42c56 node::NodeMainInstance::Run() [node /home/[...]/index.js]
| 14: 0xac89bc node::Start(int, char**) [node /home/[...]/index.js]
| 15: 0x7fc10834c850  [/usr/lib/libc.so.6]
| 16: 0x7fc10834c90a __libc_start_main [/usr/lib/libc.so.6]
| 17: 0xa3d03c  [node /home/[...]/index.js]
PM2      | App [xxx:0] exited with code [0] via signal [SIGABRT]
PM2      | App [xxx:0] starting in -fork mode-
PM2      | App [xxx:0] online
`

cleoo, 2024-09-05:
The absence of a solution will force me to opt for an alternative solution to NodeJS, which frankly upsets me greatly, but it's inconceivable to let machines crash during payment processes just because someone can't catch an exception ...

cleoo, 2024-09-10:
#809 #810 
to be continued

closed 2024-09-29 by cleoo