node-usb/node-usb #687: EXC_BAD_ACCESS (SIGSEGV), FatalProcessOutOfMemory on Big Sur 11.7.10

semireg, 2023-10-25:
## Bug Description:
Crash report from one user. Otherwise a stable Electron app. I can't reproduce. 

## Additional Information
- macOS Big Sur 11.7.10
- Electron 26.2.2
- Node USB Version 2.11.0

## Crash Report
```
Process:               APP [6196]
Path:                  /private/var/folders/*/APP.app/Contents/MacOS/APP
Identifier:            com.company.APP
Version:               A.B.C (XXXX)
Code Type:             ARM-64 (Native)
Parent Process:        ??? [1]
Responsible:           APP [6196]
User ID:               502

Date/Time:             2023-10-25 16:32:37.319 -0400
OS Version:            macOS 11.7.10 (20G1427)
Report Version:        12
Anonymous UUID:        .....

Sleep/Wake UUID:       .....

Time Awake Since Boot: 3500 seconds
Time Since Wake:       55 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [6196]

VM Regions Near 0:
--> 
    __TEXT                      100370000-100374000    [   16K] r-x/r-x SM=COW  /var/folders/*/APP.app/Contents/MacOS/APP

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.github.Electron.framework 	0x0000000100b1c374 node::FreeArrayBufferAllocator(node::ArrayBufferAllocator*) + 16788
1   com.github.Electron.framework 	0x0000000100b1c370 node::FreeArrayBufferAllocator(node::ArrayBufferAllocator*) + 16784
2   com.github.Electron.framework 	0x0000000101880bdc v8::internal::V8::FatalProcessOutOfMemory(v8::internal::Isolate*, char const*, char const*) + 96
3   com.github.Electron.framework 	0x0000000101882d5c v8::Template::Set(v8::Local<v8::Name>, v8::Local<v8::Data>, v8::PropertyAttribute) + 300
4   com.github.Electron.framework 	0x0000000106cb6d08 napi_define_class + 824
5   usb_bindings.node             	0x0000000118c92a2c Napi::ObjectWrap<Device>::DefineClass(Napi::Env, char const*, unsigned long, napi_property_descriptor const*, void*) + 432
6   usb_bindings.node             	0x0000000118c90448 Device::Init(Napi::Env, Napi::Object) + 952
7   usb_bindings.node             	0x0000000118c83d30 Init(Napi::Env, Napi::Object) + 196
8   usb_bindings.node             	0x0000000118c894dc __napi_Init(napi_env__*, napi_value__*) + 24
9   com.github.Electron.framework 	0x0000000106cbf20c node::AsyncResource::CallbackScope::CallbackScope(node::AsyncResource*) + 3032
10  com.github.Electron.framework 	0x0000000106cce110 node::binding::get_linked_module(char const*) + 3468
11  com.github.Electron.framework 	0x0000000106ca3e98 node::EmitAsyncDestroy(node::Environment*, node::async_context) + 288116
12  com.github.Electron.framework 	0x0000000106ccceac node_module_register + 2468
13  com.github.Electron.framework 	0x00000001018e8e58 v8::internal::Accessors::MakeAccessor(v8::internal::Isolate*, v8::internal::Handle<v8::internal::Name>, void (*)(v8::Local<v8::Name>, v8::PropertyCallbackInfo<v8::Value> const&), void (*)(v8::Local<v8::Name>, v8::Local<v8::Value>, v8::PropertyCallbackInfo<v8::Boolean> const&)) + 11048
```

semireg, 2023-10-27:
Another from Big Sur.

```
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           APP [1761]
User ID:               501
 
Date/Time:             2023-10-26 15:53:07.824 -0600
OS Version:            macOS 11.7.6 (20G1231)
Report Version:        12
Anonymous UUID:        8F954A0C-EA0E-68A8-5022-DED135E9B35A
 
 
Time Awake Since Boot: 4900 seconds
 
System Integrity Protection: enabled
 
Crashed Thread:        0  Dispatch queue: com.apple.main-thread
 
Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY
 
Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [1761]
 
VM Regions Near 0:
-->
    __TEXT                      10e25e000-10e260000    [    8K] r-x/r-x SM=COW  /Applications/APP.app/Contents/MacOS/APP
 
Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.github.Electron.framework           0x000000011091090b v8_inspector::V8DebuggerId::V8DebuggerId(std::Cr::pair<long long, long long>) + 24571
1   com.github.Electron.framework           0x0000000111a1c731 v8::internal::V8::FatalProcessOutOfMemory(v8::internal::Isolate*, char const*, char const*) + 81
2   com.github.Electron.framework           0x0000000111a1ed16 v8::Template::Set(v8::Local<v8::Name>, v8::Local<v8::Data>, v8::PropertyAttribute) + 310
3   com.github.Electron.framework           0x000000011772639b napi_define_class + 1003
4   usb_bindings.node                   0x0000000110181538 Napi::ObjectWrap<Device>::DefineClass(Napi::Env, char const*, unsigned long, napi_property_descriptor const*, void*) + 536
5   usb_bindings.node                   0x000000011017f055 Device::Init(Napi::Env, Napi::Object) + 1189
6   usb_bindings.node                   0x000000011017222e Init(Napi::Env, Napi::Object) + 206
7   usb_bindings.node                   0x0000000110177b21 __napi_Init(napi_env__*, napi_value__*) + 17
8   com.github.Electron.framework           0x00000001177306e8 node::AsyncResource::CallbackScope::CallbackScope(node::AsyncResource*) + 3112
9   com.github.Electron.framework           0x0000000117742ac0 node::binding::get_linked_module(char const*) + 3840
10  com.github.Electron.framework         0x000000011770e6d1 node::EmitAsyncDestroy(node::Environment*, node::async_context) + 331025
11  com.github.Electron.framework         0x0000000117741659 node_module_register + 2553
12  com.github.Electron.framework         0x0000000111a8284f v8::internal::Accessors::MakeAccessor(v8::internal::Isolate*, v8::internal::Handle<v8::internal::Name>, void (*)(v8::Local<v8::Name>, v8::PropertyCallbackInfo<v8::Value> const&), void (*)(v8::Local<v8::Name>, v8::Local<v8::Value>, v8::PropertyCallbackInfo<v8::Boolean> const&)) + 10847
13  com.github.Electron.framework         0x00000001104a7df6 0x1103de000 + 826870
14  ???                               0x000000017052903a 0 + 6179426362
15  com.github.Electron.framework         0x0000000110410ce6 0x1103de000 + 208102
16  ???                               0x00000001705293a8 0 + 6179427240
17  ???                               0x000000017050d486 0 + 6179312774
18  ???                               0x00000001704ccd56 0 + 6179048790
19  ???                               0x00000001704975f1 0 + 6178829809
20  ???                               0x0000000170497a44 0 + 6178830916
21  ???                               0x000000017049e6d2 0 + 6178858706
```

semireg, 2023-10-30:
Update: I've tried building with the latest Electron and still same crash.

thegecko, 2023-10-30:
Are you able to get this to crash using the example electron app?

https://github.com/node-usb/node-usb-example-electron

I've just updated it to the latest versions and tested it on macOS Sonoma

semireg, 2023-10-30:
I'm not sure, because I don't have access to Big Sur. This app is otherwise stable. Crash reports are only coming in from Big Sur users with the app packaged with node-usb 2.11.0. Strange...

semireg, 2023-11-02:
Update with more information, but no resolution (yet). 

My app:
- I have a copy of Big Sur running on a VM on an Intel Mac. I have reproduced my app's crash. 👍 
- I downgraded to 2.9.0 and still have the crash on Big Sur.

node-usb-example-electron:
- I ran on Big Sur and it works. It is using `WebUSB` whereas my app uses `getDeviceList()`.
- I added a call to `getDeviceList()` and it worked.

Very strange. I'll keep downgrading Electron and other deps until I get it working again. Stay tuned.

semireg, 2023-11-03:
I went through old beta builds and found a point in the builds where one build worked on Big Sur and the next crashed. I went through the git history and didn't see anything obvious. I had a hunch there was something subtle in the node API (NAPI) or ABI (node-abi)... out of sync between package.json, package-lock.json, and node_modules. I nuked node_modules, upgraded node-abi from 3.47.0 to 3.51.0, 🙏 and sure enough, it's working again. I really can't explain it... but at least we have some reference in case someone else experiences this crash on startup. Closing the issue out of sheer awkwardness and exhaustion. 😩

closed 2023-11-03 by semireg

thegecko, 2023-11-03:
Glad you found a fix! Your approach could be useful to others in future, too (albeit a big change)