node-usb/node-usb #695: Error when using electron-forge

joncamp, 2023-11-07:
## Bug Description:

Using an Electron-Forge based solution, running the solution causes errors

## Steps to Reproduce:
1. clone https://github.com/joncamp/node-usb-example-electron-forge
2. npm install
3. npm start

Result:
Debugger listening on ws://127.0.0.1:62348/f3dd7cd3-d58f-49a5-969c-01e67c9b87cc
For help, see: https://nodejs.org/en/docs/inspector
App threw an error during load
Error: No native build was found for platform=win32 arch=x64 runtime=electron abi=118 uv=1 libc=glibc node=18.17.1 electron=27.0.3 webpack=true
    loaded from: C:\Users\InsertCoolUserHere\source\repos\node-usb-example-electron-forge

    at load.resolve.load.path (C:\Users\InsertCoolUserHere\source\repos\node-usb-example-electron-forge\.webpack\main\index.js:977:9)
    at load (C:\Users\InsertCoolUserHere\source\repos\node-usb-example-electron-forge\.webpack\main\index.js:939:30)
    at ./node_modules/usb/dist/usb/bindings.js (C:\Users\InsertCoolUserHere\source\repos\node-usb-example-electron-forge\.webpack\main\index.js:1232:96)
    at __webpack_require__ (C:\Users\InsertCoolUserHere\source\repos\node-usb-example-electron-forge\.webpack\main\index.js:3006:42)
    at ./node_modules/usb/dist/usb/device.js (C:\Users\InsertCoolUserHere\source\repos\node-usb-example-electron-forge\.webpack\main\index.js:1288:13)
    at __webpack_require__ (C:\Users\InsertCoolUserHere\source\repos\node-usb-example-electron-forge\.webpack\main\index.js:3006:42)
    at ./node_modules/usb/dist/usb/index.js (C:\Users\InsertCoolUserHere\source\repos\node-usb-example-electron-forge\.webpack\main\index.js:1806:18)
    at __webpack_require__ (C:\Users\InsertCoolUserHere\source\repos\node-usb-example-electron-forge\.webpack\main\index.js:3006:42)
    at ./node_modules/usb/dist/webusb/index.js (C:\Users\InsertCoolUserHere\source\repos\node-usb-example-electron-forge\.webpack\main\index.js:2070:13)
    at __webpack_require__ (C:\Users\InsertCoolUserHere\source\repos\node-usb-example-electron-forge\.webpack\main\index.js:3006:42)

## Additional Information
This is consistent with the behavior seen in #518 but with a minimal repro. I have tested the https://github.com/node-usb/node-usb-example-electron and it works as expected. 
Versions Node.js 18.17.1, Chromium 118.0.5993.120, and Electron 27.0.3.

strazzere, 2023-12-05:
Your issue is because the `__dirname` is getting overloaded by a webpack polyfill. Here is a git diff for this to get fixed, I'll submit a PR on your example branch shortly and link this issue;

```
diff --git a/src/index.ts b/src/index.ts
index 015070a..6c62cf0 100644
--- a/src/index.ts
+++ b/src/index.ts
@@ -1,5 +1,5 @@
 import { app, BrowserWindow } from 'electron';
-import usb from 'usb';
+import { WebUSB } from 'usb';
 
 // This allows TypeScript to pick up the magic constants that's auto-generated by Forge's Webpack
 // plugin that tells the Electron app where to look for the Webpack-bundled app code (depending on
@@ -9,7 +9,7 @@ declare const MAIN_WINDOW_PRELOAD_WEBPACK_ENTRY: string;
 
 const windows: BrowserWindow[] = [];
 
-const webusb = new usb.WebUSB({
+const webusb = new WebUSB({
     allowAllDevices: true
 });
 
diff --git a/webpack.main.config.ts b/webpack.main.config.ts
index 509bbb0..c356d1a 100644
--- a/webpack.main.config.ts
+++ b/webpack.main.config.ts
@@ -9,13 +9,11 @@ export const mainConfig: Configuration = {
    */
   entry: './src/index.ts',
   // Put your normal webpack config below here
+  node: { __dirname: true },
   module: {
     rules,
   },
   resolve: {
     extensions: ['.js', '.ts', '.jsx', '.tsx', '.css', '.json'],
   }
-  // externals: {
-  //   usb: 'commonjs usb'
-  // }
 };
\ No newline at end of file
```

closed 2023-12-06 by joncamp

RobertUpchurch, 2024-04-05:
Hi, I implemented the changes as seen above and it does indeed fix the problems when running locally, however when running `make` it seems this issue re-appears when opening the app. Any ideas?

strazzere, 2024-04-05:
I've commented on your new issue, though just to leave a little extra information here -- I think you need to give more information on the error you're receiving. I'm unable to reproduce what you're seeing when running `npm run make`.