node-usb/node-usb #707: win7 Report an error using nodeusb

a37836323, 2023-12-06:
environment
  "electron": "22.3.23",
  "electron-rebuild": "^3.2.9",
     "usb": "^2.11.0",
Compile using electron-builder on windos10 64-bit
Used before compilation to rebuild
  .\node_modules\.bin\electron-rebuild.cmd

Win10 and win11 open without any problem, but win7 64-bit opens with an error:

[Window Title]
Error

[Main Instruction]
A JavaScript error occurred in the main process

[Content]
Uncaught Exception:
Error: The specified procedure could not be found.
\\?\C:\Users\Administrator\AppData\Local\Temp\07d7f508-d52b-4e17-b409-d218a396510d.tmp.node
     at process.func [as dlopen] (node:electron/js2c/asar_bundle:2:1869)
     at Module._extensions..node (node:internal/modules/cjs/loader:1226:18)
     at Object.func [as .node] (node:electron/js2c/asar_bundle:2:2096)
     at Module.load (node:internal/modules/cjs/loader:1011:32)
     at Module._load (node:internal/modules/cjs/loader:846:12)
     at f._load (node:electron/js2c/asar_bundle:2:13377)
     at Module.require (node:internal/modules/cjs/loader:1035:19)
     at require (node:internal/modules/cjs/helpers:102:18)
     at load (E:\yunwu-setup\resources\app.asar\node_modules\node-gyp-build\node-gyp-build.js:22:10)
     at Object.<anonymous> (E:\yunwu-setup\resources\app.asar\node_modules\usb\dist\usb\bindings.js:8:38)

Hope you can help me, thank you!

ajuner, 2023-12-08:
try "usb": "2.4.3"

thegecko, 2023-12-09:
> try "usb": "2.4.3"

I'd be interested to know if this does fix the issue.

I'd also recommend not rebuilding the native modules like electron wants you to do. We ship electron compliant binaries already

ajuner, 2023-12-12:
I encountered the same problem using the latest version in electron. I don’t know what the principle behind it is, but the problem was indeed solved after downgrading to  v2.4.3

ajuner, 2023-12-12:
I get it from this issue
https://github.com/node-usb/node-usb/issues/555