node-usb/node-usb #714: "transfer" of Endpoint is undefined

kemkriszt, 2023-12-29:
## Bug Description:

I can't use `transfer` using the WebUSB api but can with the legacy API

```ts
// This does work
const devices = getDeviceList()
const device = devices[0]

device.__open()
device.__claimInterface(0)
device.open()

const ifc = device.interface(0);
ifc.claim();

const outEndpoint: OutEndpoint = <OutEndpoint>ifc.endpoints[0];
console.log(outEndpoint.transfer) // function
```

```ts
// This doesn't
const customWebUsb = new WebUSB({allowAllDevices: true})
const devices = await customWebUsb.getDevices()

const device = devices[0]
device.open()
device.selectConfiguration(1)
device.claimInterface(0)

console.log(device.configuration!.interfaces[0].alternate.endpoints[0].transfer) // undefined
```

Am I missing some steps from setup?
Note that I wasn't trying to use `transfer` directly, I was trying to write some data using `transferOut` which gave me the undefined error so I did some digging and got here

## Steps to Reproduce:
1.  yarn add usb
2.  brew install libusb // however I'm not sure if this is needed. I saw the same behaviour before it as well

## Additional Information
- Operating System: macOS Sonoma
- Runtime: node 18.17
- Node USB Version:  2.11.0

thegecko, 2023-12-29:
WebUSB endpoints don't have a transfer function, see:

https://wicg.github.io/webusb/#usbendpoint

Please use `transfer` or `controlTransfer` functions on the WebUSB device:

https://wicg.github.io/webusb/#dom-usbdevice-controltransferin
https://wicg.github.io/webusb/#dom-usbdevice-controltransferout
https://wicg.github.io/webusb/#dom-usbdevice-transferin
https://wicg.github.io/webusb/#dom-usbdevice-transferout

kemkriszt, 2023-12-30:
Sorry I wasn't clear enough. I was using **transferOut** on the device not the endpoint and it gave me that error that **transfer** is undefined

In the meantime I've figured out that I wasn't opening the device where I thought I was. Would it make sense for it to throw some more meaningful error if trying to write to a device that is not opened?

thegecko, 2023-12-30:
> Would it make sense for it to throw some more meaningful error

Probably. We try to implement the WebUSB specification exactly and that check isn't mentioned. However if chrome throws an error at this point, I think it is worth copying the behaviour. Would need checking.

kemkriszt, 2023-12-30:
Yeah, good point.

I've just checked and Chrome does throw an error:
[image]

PR #715 (2024-01-04): Throw error when transferring and device closed
## Changes

- Added check in WebUSB API for device to be opened before transferring

## Fixes

- Fixes #714

## Checklist

Have you...

- [x] Tested the change acts as expected
- [x] Checked the change doesn't remove or change existing functionality
- [x] Considered how the feature impacts the product and tested around it (not just the happy paths)
- [x] Where possible, executed the hardware tests on multiple operating systems
closed 2024-01-04 by thegecko