node-usb/node-usb #717: Crash macOS 14.2.1 EXC_BAD_ACCESS on libusb_close

semireg, 2024-01-02:
This seems like a rare crash. Using latest node-usb 2.11.0 w/ electron 26.6.2.

```
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               502

Date/Time:             2024-01-02 07:23:26.1113 -0500
OS Version:            macOS 14.2.1 (23C71)
Report Version:        12

Time Awake Since Boot: 3300 seconds
Time Since Wake:       2510 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000090
Exception Codes:       0x0000000000000001, 0x0000000000000090

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [1261]

VM Region Info: 0x90 is not in any region.  Bytes before following region: 57011261669232
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      Memory Tag 255           33d9f8000000-33da00000000 [128.0M] ---/rwx SM=NUL  

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   usb_bindings.node             	       0x102839da8 do_close + 120
1   usb_bindings.node             	       0x102839c9c libusb_close + 184
2   usb_bindings.node             	       0x10282bf50 Device::~Device() + 256
3   usb_bindings.node             	       0x10282bfcc Device::~Device() + 12
4   usb_bindings.node             	       0x102832100 Napi::ObjectWrap<Device>::FinalizeCallback(napi_env__*, void*, void*) + 80
5   Electron Framework            	       0x11a71b2f0 node_api_get_module_file_name + 6920
6   Electron Framework            	       0x11a717244 node::AsyncResource::CallbackScope::CallbackScope(node::AsyncResource*) + 1752
7   Electron Framework            	       0x11a70eb24 node::Stop(node::Environment*, node::StopFlags::Flags) + 3684
8   Electron Framework            	       0x11a716de8 node::AsyncResource::CallbackScope::CallbackScope(node::AsyncResource*) + 636
9   Electron Framework            	       0x11a71a864 node_api_get_module_file_name + 4220
10  Electron Framework            	       0x11a71a8bc node_api_get_module_file_name + 4308
11  Electron Framework            	       0x11a71abb4 node_api_get_module_file_name + 5068
12  Electron Framework            	       0x1142f0a18 uv_run + 732
13  Electron Framework            	       0x11a6fe754 node::EmitAsyncDestroy(node::Environment*, node::async_context) + 300312
14  Electron Framework            	       0x11a6fee38 node::EmitAsyncDestroy(node::Environment*, node::async_context) + 302076
15  Electron Framework            	       0x11a6b2e68 node::FreeEnvironment(node::Environment*) + 148
16  Electron Framework            	       0x1143fc080 v8::internal::HeapAllocator::HeapAllocator(v8::internal::Heap*) + 36
17  Electron Framework            	       0x1143e518c v8::internal::compiler::SLVerifierHintParametersOf(v8::internal::compiler::Operator const*) + 68796
18  Electron Framework            	       0x11632da20 v8::internal::SetupIsolateDelegate::SetupBuiltins(v8::internal::Isolate*, bool) + 3790472
19  Electron Framework            	       0x11632f1ac v8::internal::SetupIsolateDelegate::SetupBuiltins(v8::internal::Isolate*, bool) + 3796500
20  Electron Framework            	       0x11632b4f0 v8::internal::SetupIsolateDelegate::SetupBuiltins(v8::internal::Isolate*, bool) + 3780952
21  Electron Framework            	       0x1145df11c v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 13728
22  Electron Framework            	       0x1145e0038 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 17596
23  Electron Framework            	       0x1145dfe88 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 17164
24  Electron Framework            	       0x1145dea08 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 11916
25  Electron Framework            	       0x1145deb18 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 12188
26  Electron Framework            	       0x114301c10 ElectronMain + 128
27  dyld                          	       0x186bb90e0 start + 2360
```

thegecko, 2024-01-02:
Would be really useful if this could be replicated without being run in electron.

semireg, 2024-01-02:
Agreed. These usually come in from user reports and I have limited knowledge of how they're triggering the crash. It seemed isolated and non reproducible from their side. They said it happened when they were updating the app, so that would mean the app was shutting down and it crashed on an already terminating app/process. Just an idea. Feel free to close as non-reproducible and I'll reopen if I get more info. 👍

thegecko, 2024-01-02:
> Feel free to close as non-reproducible and I'll reopen if I get more info. 

Thanks, will do

closed 2024-01-02 by thegecko