node-usb/node-usb #718: How do I comply with LGPL license?

lukefwilson, 2024-01-04:
Hello,

I'd like to use this library in a proprietary application under a different license. From the documentation:

> Note that the compiled Node extension includes [libusb](https://github.com/libusb/libusb), and is thus subject to the [LGPL](https://github.com/libusb/libusb/blob/master/COPYING).

From my understanding the best way to comply with LGPL is to be able to dynamically link the libusb library, but that doesn't seem possible if it's being compiled statically in the Node extension. 

Does anyone have any advice on how to use this library while complying with libusb's LGPL license?

Thanks in advance!

thegecko, 2024-01-04:
I'm no legal expert, so the usual "this is not legal advice" applies, however from what I read:

- [LGPL](https://fossa.com/blog/open-source-software-licenses-101-lgpl-license/) does not require distribution of source if the source is not changed. I'd argue we don't change libusb and simply compile it and ship it as is with node-usb.
- As a backup (and to cover copyright), I'd recommend you reference this repository in your third party IP declaration or similar

I'm sure others may have more/different/better advice, too.

kevinmehall, 2024-01-04:
LGPL requires that end users must be able to recompile and replace the LGPL code with their own modified version. Dynamic linking is often suggested as the best way to achieve this, because the user can compile their own and swap out the library file. Assuming you don't modify the node-usb C++ code in this repo, or treat it as if it were LGPL too and distribute any changes to it, a user could compile node-usb against the modified libusb. The `.node` extension file is a dynamic library that can be recompiled independently of the rest of your app. You don't want to statically link LGPL code with proprietary code, because a user can't recompile the proprietary code, but linking it with code under a compatible open source license (MIT) is fine if you treat the combined library as LGPL.

Probably node-usb should have just been LGPL itself to avoid this confusion, but MIT adds a little bit of flexibility, if hypothetically its use of libusb were replaced with [nusb](https://github.com/kevinmehall/nusb) or something not LGPL in the future.

lukefwilson, 2024-01-05:
Thank you both, this is super helpful! 

For those who are curious, by default `electron-builder` packages up all .node extensions into an app.asar archive, so you won't be able to find the usb.node dynamic library in your builds (which makes it rather hard to replace in an LGPL-friendly way). However, it's an easy config change to have these extension files kept outside of the archive so that they are truly replaceable. Just add this `electron-builder` config to your `package.json`:

```
"build": {
        "asar": true,
        "asarUnpack": [
            "**/*.node"
        ]
    }
```

closed 2024-01-05 by lukefwilson