node-usb/node-usb #722: Android libudev problem

tomek3e, 2024-01-11:
## Bug Description:
Trying to run on Android, getting error:
`Error dlopen failed: library "libudev.so.1"`
As far as I know there are no `libudev` ports for Android.

## Additional Information
- Operating System: Android (Emteria Android with Termux)
- Runtime: node v20.02.0
- Node USB Version (`npm ls usb`): 2.11.0

thegecko, 2024-01-20:
I've never tested this on android TBF, anyone else ever got it to work?

strazzere, 2024-01-29:
This is an upstream problem with termux, it doesn't support `libudev` when compiling `libusb` [1]. You'll need to start by getting that to work before the current project will work.

1 - https://github.com/termux/termux-packages/blob/master/packages/libusb/build.sh

closed 2024-06-27 by tomek3e