node-usb/node-usb #743: 2.12.0 prebuilds seem to have an error getting picked up

strazzere, 2024-03-06:
I'm not positive what is going on here as I'm not overly familiar with nodegyp, etc. However I noticed when bumping from `2.11.0` to `2.12.0` the `linux-x86` prebuild not longer is properly detected. This results in everything being build on a few of my CI machines. Currently, it is throwing as I did not have `libusb-dev` installed. Obviously, this is something I can get around by changing those images to have this included, though I wanted to highlight it as a prebuilt regression.

On a curourly look, I notices all the prebuild output filenames changed;

```
find . -name '*.node'                                                                                        ⇣22.4 KiB/s ⇡3.16 KiB/s 192.168.1.136   98.97.58.71   99%   270 Mbps  
./2.11.0/package/prebuilds/linux-ia32/node.napi.node
./2.11.0/package/prebuilds/win32-x64/node.napi.node
./2.11.0/package/prebuilds/linux-arm/node.napi.armv7.node
./2.11.0/package/prebuilds/linux-arm/node.napi.armv6.node
./2.11.0/package/prebuilds/linux-x64/node.napi.glibc.node
./2.11.0/package/prebuilds/linux-x64/node.napi.musl.node
./2.11.0/package/prebuilds/win32-ia32/node.napi.node
./2.11.0/package/prebuilds/android-arm/node.napi.armv7.node
./2.11.0/package/prebuilds/android-arm64/node.napi.armv8.node
./2.11.0/package/prebuilds/darwin-x64+arm64/node.napi.node
./2.11.0/package/prebuilds/linux-arm64/node.napi.armv8.node
./2.12.0/package/prebuilds/linux-ia32/usb.node
./2.12.0/package/prebuilds/win32-x64/usb.node
./2.12.0/package/prebuilds/linux-arm/usb.armv7.node
./2.12.0/package/prebuilds/linux-arm/usb.armv6.node
./2.12.0/package/prebuilds/win32-arm64/usb.node
./2.12.0/package/prebuilds/linux-x64/usb.glibc.node
./2.12.0/package/prebuilds/linux-x64/usb.musl.node
./2.12.0/package/prebuilds/win32-ia32/usb.node
./2.12.0/package/prebuilds/android-arm/usb.armv7.node
./2.12.0/package/prebuilds/android-arm64/usb.armv8.node
./2.12.0/package/prebuilds/darwin-x64+arm64/usb.node
./2.12.0/package/prebuilds/linux-arm64/usb.armv8.node
```

Will continue to dig deeper, I haven't seen anything in a diff between the tags `2.11.0` and `2.12.0` which seemingly would even cause the above change.

strazzere, 2024-03-06:
https://github.com/prebuild/prebuildify/commit/7b6dcbd0860a82db8804079ba55663affd9ae555

The above commit explains the naming change, though it's still unclear to me if this is causing the prebuilt mis-detection.

thegecko, 2024-03-06:
Thanks for looking into this, please continue to post as you find anything. I'll take a look when I can

strazzere, 2024-03-06:
Bingo;

```
Welcome to Node.js v18.14.2.
Type ".help" for more information.
> function parseTags (file) {
...   var arr = file.split('.')
...   var extension = arr.pop()
...   var tags = { file: file, specificity: 0 }
... 
...   if (extension !== 'node') return
... 
...   for (var i = 0; i < arr.length; i++) {
...     var tag = arr[i]
... 
...     if (tag === 'node' || tag === 'electron' || tag === 'node-webkit') {
...       tags.runtime = tag
...     } else if (tag === 'napi') {
...       tags.napi = true
...     } else if (tag.slice(0, 3) === 'abi') {
...       tags.abi = tag.slice(3)
...     } else if (tag.slice(0, 2) === 'uv') {
...       tags.uv = tag.slice(2)
...     } else if (tag.slice(0, 4) === 'armv') {
...       tags.armv = tag.slice(4)
...     } else if (tag === 'glibc' || tag === 'musl') {
...       tags.libc = tag
...     } else {
...       continue
...     }
... 
...     tags.specificity++
...   }
... 
...   return tags
... }
undefined
> parseTags('./2.11.0/package/prebuilds/linux-ia32/node.napi.node')
{
  file: './2.11.0/package/prebuilds/linux-ia32/node.napi.node',
  specificity: 1,
  napi: true
}
> parseTags('./2.11.0/package/prebuilds/linux-x64/node.napi.glibc.node')
{
  file: './2.11.0/package/prebuilds/linux-x64/node.napi.glibc.node',
  specificity: 2,
  napi: true,
  libc: 'glibc'
}
> parseTags('./2.12.0/package/prebuilds/linux-x64/usb.glibc.node')
{
  file: './2.12.0/package/prebuilds/linux-x64/usb.glibc.node',
  specificity: 1,
  libc: 'glibc'
}
```

So the issue is the `napi` isn't appended anymore via `prebuildify` (because it is meant to be the normal thing?) however the `node-gyp-build` (https://github.com/prebuild/node-gyp-build) expects it.

This is an easy-ish fix, but also complex. I think `prebuildify` will want to continue being this way but probably should provide some docs for it. `node-gyp-build` should likely get updated to anticipate the expectation of `napi` being the default.

Specifically to this repo, I think we could pass an expected name for the prebuild output -- thoughts?

thegecko, 2024-03-06:
> Specifically to this repo, I think we could pass an expected name for the prebuild output

That sounds like a better solution than rolling back one of the deps. Are you aware of a way to do that?

strazzere, 2024-03-07:
Seems like you could overload the `name` tag, though this doesn't feel right;

`prebuildify --napi --strip -n node.napi.usb` will output `node.napi.usb.${abi}.node` which correctly is detected.

Though I did tag the `prebuildify` project as it seemed like this was either an announced breakage, or a bug. I'm leaning towards the latter since the commit that changed it didn't mention anything about causing this. It was only meant (seemingly) to allow the package name to be included;

https://github.com/prebuild/prebuildify/commit/7b6dcbd0860a82db8804079ba55663affd9ae555#diff-e727e4bdf3657fd1d798edcd6b099d6e092f8573cba266154583a746bba0f346L114-L116

strazzere, 2024-03-07:
This will all get fixed when https://github.com/prebuild/prebuildify/pull/80 is merged and we pull a fresh release for the dependency.

Note the expected output of the prebuilds will change to something like this;
```
prebuilds/linux-x64/usb.node.napi.node
```
The `usb` package name will prepend them, but they will include the `runtime` and `napi` tags again -- and anything else we specify in the flags.

thegecko, 2024-03-07:
Awesome work @strazzere thanks for looking into this and proposing a fix upstream

thegecko, 2024-04-13:
This was fixed in #744

closed 2024-04-13 by thegecko

thegecko, 2024-05-02:
FYI, this has been re-fixed by forcing the name in #765

strazzere, 2024-05-02:
Awesome, thanks!