node-usb/node-usb #749: controlTransfer Linux always returns LIBUSB_ERROR_IO where MIDI 2.0 USB is in use

starfishmod, 2024-03-18:
## Bug Description:
I am working with MIDI 2.0 USB Devices. MIDI 2.0 gets Group Terminal Blocks using Control Transfer.
When attempting this on Linux I receive a LIBUSB_ERROR_IO message. In OSX this same code works well.

## Steps to Reproduce:
1. Run an update Linux Kernal and ALSA with MIDI 2.0 Support and connect a MIDI 2.0 USB Device. You can create a Linux Gadget to achieve this.
2. Find the device and open
3. do the following :
```js
dev.controlTransfer(0x81, 0x06, 0x2601, interfaceId, 0x5,
                (error, data) => {
                    if(error){
                       //  This will always error with LIBUSB_ERROR_IO on Linux where driver is claimed by MIDI 2.0 Driver
                    }
});
```
I believe this is/was an issue in libUSB that has been resolved(?)

Issue does not occur in OSX. Does not occur in Linux where I detach the driver and reattach manually.

## Additional Information
- Operating System: RPI running kernel 6.2
- Runtime (`node -v` or `electron -v`): electron 19.1.9
- Node USB Version (`npm ls usb`): 2.12.1

starfishmod, 2024-03-18:
> As you are on Linux, I assume your udev rules are fine?

Yes I believe so: 
```
groupadd usbusers
usermod -a -G usbusers $USER
echo 'SUBSYSTEM=="usb", MODE="0666", GROUP="usbusers"' | tee /etc/udev/rules.d/99-usbusers.rules
udevadm control --reload
udevadm trigger
```

On other systems where I dettch from the kernel the controlTransfer works.

> Any more detail on this? e.g. a PR or fixed issue we can check has made it into the release being used by node-usb.

Sorry I did read a bunch of tickets for libUSB to find what I thought had been fixed in 1.0.27 however it seems that they are windows related.

thegecko, 2024-03-18:
I don't know a lot about this, but I wonder if the kernel is taking control of the device as it has a driver installed?

mildsunrise, 2024-04-29:
IIRC, detaching the kernel driver is successful, but even then, you still get LIBUSB_ERROR_IO on transfers?

starfishmod, 2024-04-30:
> IIUC, detaching the kernel driver is successful, but even then, you still get LIBUSB_ERROR_IO on transfers?
I can dettach from the kernel successfully. If I then do a ControlTransfer it is successful.

If I do not dettach from from the Kernel on OSX, and then attempt a Control Transfer it is successfull.
If I do not dettach from from the Kernel on Linux, and then attempt a Control Transfer it is not successfull. :)
I hope I am clear?

mildsunrise, 2024-04-30:
hmm, so your point is that control transfers should work even if you don't detach the kernel driver? I'm not sure about that, it looks reasonable to me that the kernel driver reserves the device for itself.

in any case, I doubt this issue has anything to do with node-usb, at most it would be relevant to libusb