node-usb/node-usb #752: Unable to listen for USB device removal behavior connected before program startup

li725-ll, 2024-03-21:
## Bug Description:
Unable to listen for USB device removal behavior connected before program startup.

```js
const { WebUSB } = require("usb");
const usbDevice = new WebUSB({ allowAllDevices: true });

usbDevice.onconnect = () => {
    console.log("connect")
}

usbDevice.ondisconnect = () => {
    console.log("disconnect")
}
```

Firstly, I have a USB device connected to the computer, and then I start this code. After the program runs successfully, I remove the USB device connected to the computer, but it does not trigger the ondisconnect event. I found that it can only listen to USB insertion and removal events after the program is running. May I ask if this is a reasonable behavior.

## Additional Information
- Operating System: Windows 10
- Runtime (`node -v` or `electron -v`): node 16.16.0
- Node USB Version (`npm ls usb`): 2.11.0

li725-ll, 2024-03-22:
I have discovered some new things。
```js
const { WebUSB } = require("usb");

(async () => {
    const usbDevice = new WebUSB({ allowAllDevices: true });

    usbDevice.addEventListener("connect", () => {
        console.log("connect")
    })

    const usbDevices = await usbDevice.getDevices();
    console.log(usbDevices);

    usbDevice.addEventListener("disconnect", () => {
        console.log("disconnect")
    })
})()
```
If I have called the getDevices function, it will work as expected. Therefore, I believe that generally speaking, this will not affect the majority of developers using this library.If you think this issue is not important, you can close it directly. Thank you.

thegecko, 2024-04-13:
The problem here is that a [webusb disconnect event](https://wicg.github.io/webusb/#disconnect) needs to pass the webusb device being disconnected.

If the devices haven't yet been enumerated, its not possible to build the webusb device (you have to open it to get things like serialnumber and its no longer connected!)

As you have found, a workaround is to enumerate devices beforehand. We decided to not do this by default as it adds overhead.

Another workaround is to use the legacy non-webusb API which doesn't need to open the device

closed 2024-04-13 by thegecko