node-usb/node-usb #757: Electron Forge Fails To Package with node-usb

RobertUpchurch, 2024-04-05:
## Bug Description:
As seen in this #695, when you create a simple reproduction with electron forge it immedieately fails to build stating that No native build was found for platform... (see the issue for full details)

In response to that issue the "solution" was to add ```node: { __dirname: true },``` to the webpack config.

This closed the issue because it seemed to work when running locally, however if you run `yarn make` then try to open the distributable, you receive the same error. 

## Steps to Reproduce:
1. clone the reproduction https://github.com/joncamp/node-usb-example-electron-forge
2. yarn install
3. yarn make
4. open the new zip

## Additional Information
- Operating System: MacOS Sanoma
- Runtime (`node -v` or `electron -v`): node v20.12.0 - electron v29.1.4
- Node USB Version (`npm ls usb`): usb@2.12.1

strazzere, 2024-04-05:
Can you explain more about the error you're seeing? I'm unable to reproduce using `HEAD` from `https://github.com/joncamp/node-usb-example-electron-forge` - which I submitted the PR to with the fix.

I didn't utilize yarn, but here is my output;
```
$ npm run make

> node-usb-example-electron-forge@1.0.0 make
> electron-forge make

✔ Checking your system
✔ Loading configuration
✔ Resolving make targets
  › Making for the following targets: rpm, deb
✔ Running package command
  ✔ Preparing to package application
  ✔ Running packaging hooks
    ✔ Running generateAssets hook
    ✔ Running prePackage hook
      ✔ [plugin-webpack] Preparing native dependencies: 1 / 1
      ✔ [plugin-webpack] Building webpack bundles
  ✔ Packaging application
    ✔ Packaging for x64 on linux [2s]
  ✔ Running postPackage hook
✔ Running preMake hook
✔ Making distributables
  ✔ Making a rpm distributable for linux/x64 [52s]
  ✔ Making a deb distributable for linux/x64 [1m56s]
✔ Running postMake hook
  › Artifacts available at: /home/diff/repo/scratch/node-usb-example-electron-forge/out/make
```

Just to retry closer to your env, I performed the same using yarn and it also succeeded.
```
yarn make                                                          ⇣1.44 MiB/s ⇡53.1 KiB/s 192.168.1.140   98.97.59.101   98%   180 Mbps  
yarn run v1.22.10
$ electron-forge make
✔ Checking your system
✔ Loading configuration
✔ Resolving make targets
  › Making for the following targets: rpm, deb
✔ Running package command
  ✔ Preparing to package application
  ✔ Running packaging hooks
    ✔ Running generateAssets hook
    ✔ Running prePackage hook
      ✔ [plugin-webpack] Preparing native dependencies: 1 / 1
      ✔ [plugin-webpack] Building webpack bundles
  ✔ Packaging application
    ✔ Packaging for x64 on linux [1s]
  ✔ Running postPackage hook
✔ Running preMake hook
✔ Making distributables
  ✔ Making a rpm distributable for linux/x64 [50s]
  ✔ Making a deb distributable for linux/x64 [1m56s]
✔ Running postMake hook
  › Artifacts available at: /home/diff/repo/scratch/test-forge/out/make
Done in 183.32s.
```

Can you post the full stack of the error? Just because it says a similar first line, does not mean it is the same underlying polyfill issue.

RobertUpchurch, 2024-04-08:
Sure i will post shortly, to be clear the issue does not happen until you open the app from the zip file

RobertUpchurch, 2024-04-08:
Here is the output I am seeing with a successful build. Thanks for the reply!

```
✔ Checking your system
✔ Loading configuration
✔ Resolving make targets
  › Making for the following targets: zip
✔ Running package command
  ✔ Preparing to package application
  ✔ Running packaging hooks
    ✔ Running generateAssets hook
    ✔ Running prePackage hook
      ✔ [plugin-webpack] Preparing native dependencies: 1 / 1
      ✔ [plugin-webpack] Building webpack bundles
  ✔ Packaging application
    ✔ Packaging for arm64 on darwin [1s]
  ✔ Running postPackage hook
✔ Running preMake hook
✔ Making distributables
  ✔ Making a zip distributable for darwin/arm64 [7s]
✔ Running postMake hook
  › Artifacts available at: /...
```
  
  Correctly creates the zip archive, extract the zip, then open the app. 
  
  Here is the error that is thrown.
  
```
Uncaught Exception:
Error: No native build was found for platform=darwin arch=arm64 runtime=electron abi=118 uv=1 armv=8 libc=glibc node=18.17.1 electron=27.0.3 webpack=true
loaded from: /node_modules/usb

at load.resolve.load.path (/Users/rob/Documents/node-usb-example-electron-forge/out/make/zip/darwin/arm64/node-usb-example-electron-forge.app/Contents/Resources/app.asar/.webpack/main/index.js:1:10625)
at load (/Users/rob/Documents/node-usb-example-electron-forge/out/make/zip/darwin/arm64/node-usb-example-electron-forge.app/Contents/Resources/app.asar/.webpack/main/index.js:1:8118)
at 959 (/Users/rob/Documents/node-usb-example-electron-forge/out/make/zip/darwin/arm64/node-usb-example-electron-forge.app/Contents/Resources/app.asar/.webpack/main/index.js:1:12830)
at __webpack_require__ (/Users/rob/Documents/node-usb-example-electron-forge/out/make/zip/darwin/arm64/node-usb-example-electron-forge.app/Contents/Resources/app.asar/.webpack/main/index.js:1:38002)
at 479 (/Users/rob/Documents/node-usb-example-electron-forge/out/make/zip/darwin/arm64/node-usb-example-electron-forge.app/Contents/Resources/app.asar/.webpack/main/index.js:1:13441)
at __webpack_require__ (/Users/rob/Documents/node-usb-example-electron-forge/out/make/zip/darwin/arm64/node-usb-example-electron-forge.app/Contents/Resources/app.asar/.webpack/main/index.js:1:38002)
at 509 (/Users/rob/Documents/node-usb-example-electron-forge/out/make/zip/darwin/arm64/node-usb-example-electron-forge.app/Contents/Resources/app.asar/.webpack/main/index.js:1:19497)
at __webpack_require__ (/Users/rob/Documents/node-usb-example-electron-forge/out/make/zip/darwin/arm64/node-usb-example-electron-forge.app/Contents/Resources/app.asar/.webpack/main/index.js:1:38002)
at 313 (/Users/rob/Documents/node-usb-example-electron-forge/out/make/zip/darwin/arm64/node-usb-example-electron-forge.app/Contents/Resources/app.asar/.webpack/main/index.js:1:22440)
at __webpack_require__ (/Users/rob/Documents/node-usb-example-electron-forge/out/make/zip/darwin/arm64/node-usb-example-electron-forge.app/Contents/Resources/app.asar/.webpack/main/index.js:1:38002)
```

thegecko, 2024-04-13:
As we ship napi versions of the native extensions (including the one you need AFAICT), can you suppress electron forge from trying to build it?

RobertUpchurch, 2024-04-23:
Let me poke around the docs and see if theres any info on this. Do you know how to get electron forge to do that?

thegecko, 2024-04-23:
I've never used electron forge unfortunately