node-usb/node-usb #762: null-pointer access segfault in GetAllDeviceDescriptors

sfoster1, 2024-04-26:
## Bug Description:
We have an electron app that uses node-usb to listen for connect events and grab descriptors of connected devices. We've noticed that sometimes, with some devices connected (in-progress to narrow down a good example of which devices and dump their descriptor tables), our app will crash with a dump that looks like this:

```
Exception Type: EXC_BAD_ACCESS (SIGSEGV)
Exception Codes: KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes: 0x0000000000000001, 0x0000000000000000
Termination Reason: Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process: exc handler [13927]
VM Region Info: 0 is not in any region. Bytes before following region: 4338274304
REGION TYPE START - END [ VSIZE] PRT/MAX SHRMOD
REGION DETAIL
UNUSED SPACE AT START
--->
__TEXT 10294d000-10294f000 [ 8K] r-x/r-x
SM=COW /Applications/Opentrons-OT3.app/Contents/MacOS/Opentrons-OT3
Error Formulating Crash Report:
PC register does not match crashing frame (0x0 vs 0x7FF809565885)
Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0 libsystem_platform.dylib 0x7ff809565885
_platform_memmove$VARIANT$Rosetta + 69
1 Electron Framework 0x11c61de99
node::Buffer::Copy(v8::Isolate*, char const*, unsigned long) + 361
2 Electron Framework 0x11c61dd7b
node::Buffer::Copy(v8::Isolate*, char const*, unsigned long) + 75
3 Electron Framework 0x11c6043f6 napi_create_buffer_copy +
262
4 usb_bindings.node 0x10beaa512 Device::cdesc2V8(Napi::Env,
libusb_config_descriptor*) + 930
5 usb_bindings.node 0x10beabb3b
Device::GetAllConfigDescriptors(Napi::CallbackInfo const&) + 139
6 usb_bindings.node 0x10beb0fa0
Napi::InstanceWrap<Device>::InstanceMethodCallbackWrapper(napi_env__*,
napi_callback_info__*)::'lambda'()::operator()() const + 144
7 usb_bindings.node 0x10beb0eba
Napi::InstanceWrap<Device>::InstanceMethodCallbackWrapper(napi_env__*,
napi_callback_info__*) + 42
8 Electron Framework 0x11c60159c napi_is_detached_arraybuffer
+ 268
9 Electron Framework 0x114d70e03 0x114d3b000 + 220675
```
(and so on...)

Poking around in the source, since it can be a bit challenging to get a debugger with symbols connected here, I notice something about the crash area: I don't think node-usb is handling possible null pointers in the `extra` field of the endpoint and interface descriptors (I actually don't think `libusb` is either, but maybe that's a later problem).

Specifically, `libusb_config_descriptor`, `libusb_endpoint_descriptor`, and `libusb_interface_descriptor` all have this `const unsigned char * extra` element that's supposed to hold remaining unparsed data after the standard data has been parsed out. `Device::cdesc2V8` will copy those into corresponding object keys for the node descriptor with calls to `DefineProperty` for the [config](https://github.com/node-usb/node-usb/blob/8813a2d1dc38e00dbdfbf2559ad92179d474daa2/src/device.cc#L108) , [interface](https://github.com/node-usb/node-usb/blob/8813a2d1dc38e00dbdfbf2559ad92179d474daa2/src/device.cc#L136), and [endpoint](https://github.com/node-usb/node-usb/blob/8a41bebf7b3cc57fa2f3845107679fa01b70c4ba/src/device.cc#L155) descriptors.

The thing is that libusb actually has several cases where those values in the descriptors that libusb returns can actually be null, if I'm reading this right. 

In the configuration descriptor,
- the descriptor base is `calloc`'d [before `parse_configuration`](https://github.com/libusb/libusb/blob/f8a6c412f51c974d8c70fd100e28b3b69b9d3318/libusb/descriptor.c#L461) so `extra` is initialized to `NULL`
- `extra` is `realloc`'d [here](https://github.com/libusb/libusb/blob/f8a6c412f51c974d8c70fd100e28b3b69b9d3318/libusb/descriptor.c#L426), but this only happens _if there's actually unparsed data_, and there might not be, and that's a valid case

In the interface descriptor,
- each altsetting is `realloc`'d [here](https://github.com/libusb/libusb/blob/f8a6c412f51c974d8c70fd100e28b3b69b9d3318/libusb/descriptor.c#L206) and `extra` is explicitly `NULL`-initialized [here](https://github.com/libusb/libusb/blob/f8a6c412f51c974d8c70fd100e28b3b69b9d3318/libusb/descriptor.c#L236)
- the `extra` assignment is similarly guarded by an `if` for if there's unparsed data [here](https://github.com/libusb/libusb/blob/f8a6c412f51c974d8c70fd100e28b3b69b9d3318/libusb/descriptor.c#L282)

In the endpoint descriptor,
- there's an early return if there's no extra data [here](https://github.com/libusb/libusb/blob/f8a6c412f51c974d8c70fd100e28b3b69b9d3318/libusb/descriptor.c#L152) which leaves `->extra` NULL

In all of these cases, there will be a call like this in `cdesc2V8`: `Buffer::Copy(env, (const char*)NULL, 0)`. Even though there's a 0 size, this is definitely UB, and I'm pretty sure that in practice at least the rosetta arm `memmove` is faulting on it per the crash dump above - the dest pointer [appears to be checked](https://github.com/nodejs/node/blob/1d455bc0fdbaa6980590b0176f5c1645018bdf82/src/node_buffer.cc#L417-L420) inside node source.

I _think_ the fix for this is null-checking the extra field and putting a `Buffer::New()` into the property in `cdesc2V8` if the extra descriptor is `NULL`.. If that sounds good to you, I can open a PR. Unfortunately (see below) this is kind of a pain to reproduce!

## Steps to Reproduce:
There is not a clear reproduction case unfortunately, but the crashdump above (full dump attached) should help.

This happened on a Model: MacBookPro17,1, BootROM 10151.101.3, proc 8:4:4 processors, 16 GB, SMC with SIP on with a Tripp-Lite
[U444-06N-H4GU-C](https://tripplite.eaton.com/usb-3-1-gen-1-usb-c-to-hdmi-4k-adapter-usb-a-usb-c-pd-charging-gigabit-ethernet-thunderbolt-3-compatible-4kx2k~U44406NH4GUC) (I think, these are all remarkably similar) and a USB mouse attached. Sometimes. I'd bet it's a race condition.

## Additional Information
- Operating System: macOS 14.4.1 (23E224)
- Runtime: electron 27.0.0
- Node USB Version (`npm ls usb`): 2.12.0

Crash dump:
[node-usb-crash.txt](https://github.com/node-usb/node-usb/files/15135405/node-usb-crash.txt)

thegecko, 2024-04-29:
Thanks for your detailed report @sfoster1 .

Would you mind creating a PR with these fixes and we can see if it fixes your issue.

sfoster1, 2024-04-29:
> Thanks for your detailed report @sfoster1 .
> 
> Would you mind creating a PR with these fixes and we can see if it fixes your issue.

#764 is the PR, will be testing today

PR #764 (2024-05-02): Nullchecks and errorchecks in descriptor parsing
There were a couple places in device.cc where libusb could fail or could return nulls that weren't checked; this PR adds checks there.

## Changes
Specifically,
- libusb_get_config_descriptor() is nonblocking but it allocates and checks that allocation, so we need to check its return value
- the ->extra fields in config, interface, and endpoint descriptors, which gather unparsed data, can sometimes be returned as NULL. The ->extra_length is always 0 in such cases, but calling memmove(dest, NULL, 0) as happens inside the eventual napi buffer copi [... cut]
commit ae599fee (2024-05-02): Nullchecks and errorchecks in descriptor parsing (#764)

There were a couple places in device.cc where libusb could fail or could
return nulls that weren't checked; this PR adds checks there.

Specifically,
- libusb_get_config_descriptor() is nonblocking but it allocates and
checks that allocation, so we need to check its return value
- the ->extra fields in config, interface, and endpoint descriptors,
which gather unparsed data, can sometimes be returned as NULL. The
->extra_length is always 0 in such cases, but calling memmove(dest,
NULL, 0) as happens inside the eventual napi buffer copies  [... cut]
closed 2024-05-02 by thegecko