node-usb/node-usb #763: libudev.h: No such file or directory

mateja176, 2024-04-29:
## Bug Description:

## Steps to Reproduce:
1. npm i usb

## Additional Information
- Operating System: Debian 12 Bookworm
- Runtime (`node -v` or `electron -v`): node v20.11.1 (lts)
- Node USB Version (`npm ls usb`): 2.12.1

thegecko, 2024-04-29:
Does it try to build node-usb or is this a runtime error?

mateja176, 2024-04-29:
> Does it try to build node-usb or is this a runtime error?

The error occurs during build time, before the package can be installed.

mateja176, 2024-04-29:
@thegecko Here's the complete `error.log`:
```shell
error /home/user/node_modules/usb: Command failed.
Exit code: 1
Command: node-gyp-build
Arguments:
Directory: /home/user/node_modules/usb
Output:
gyp info it worked if it ends with ok
gyp info using node-gyp@10.0.1
gyp info using node@20.11.1 | linux | x64
gyp info find Python using Python version 3.11.8 found at "/home/user/.nix-profile/bin/python3"

gyp info spawn /home/user/.nix-profile/bin/python3
gyp info spawn args [
gyp info spawn args '/home/user/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args 'binding.gyp',
gyp info spawn args '-f',
gyp info spawn args 'make',
gyp info spawn args '-I',
gyp info spawn args '/home/user/node_modules/usb/build/config.gypi',
gyp info spawn args '-I',
gyp info spawn args '/home/user/node_modules/node-gyp/addon.gypi',
gyp info spawn args '-I',
gyp info spawn args '/home/user/.cache/node-gyp/20.11.1/include/node/common.gypi',
gyp info spawn args '-Dlibrary=shared_library',
gyp info spawn args '-Dvisibility=default',
gyp info spawn args '-Dnode_root_dir=/home/user/.cache/node-gyp/20.11.1',
gyp info spawn args '-Dnode_gyp_dir=/home/user/node_modules/node-gyp',
gyp info spawn args '-Dnode_lib_file=/home/user/.cache/node-gyp/20.11.1/<(target_arch)/node.lib',
gyp info spawn args '-Dmodule_root_dir=/home/user/node_modules/usb',
gyp info spawn args '-Dnode_engine=v8',
gyp info spawn args '--depth=.',
gyp info spawn args '--no-parallel',
gyp info spawn args '--generator-output',
gyp info spawn args 'build',
gyp info spawn args '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/home/user/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/events_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
../libusb/libusb/os/linux_udev.c:28:10: fatal error: libudev.h: No such file or directory
   28 | #include <libudev.h>
      |          ^~~~~~~~~~~
compilation terminated.
make: *** [libusb.target.mk:149: Release/obj.target/libusb/libusb/libusb/os/linux_udev.o] Error 1
make: Leaving directory '/home/user/node_modules/usb/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack at ChildProcess.<anonymous> (/home/user/node_modules/node-gyp/lib/build.js:209:23)
gyp ERR! System Linux 6.1.0-20-amd64
gyp ERR! command "/nix/store/0akvkk9k1a7z5vjp34yz6dr91j776jhv-nodejs-20.11.1/bin/node" "/home/user/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/user/node_modules/usb

```

thegecko, 2024-04-29:
Hmm, this looks like a missing file on your machine, but your packages look right. What's concerning me more is why it is trying to build at all. We ship the binaries for linux x86_64 so it shouldn't be building at all!

https://github.com/node-usb/node-usb/releases/tag/v2.12.1

mildsunrise, 2024-04-29:
yes, exactly. I'd like to see the error that is causing the prebuild to fail loading (and thus triggering the fallback build), please do:

``` bash
npm install --ignore-scripts usb
node -p 'require("usb")'
```

on the other hand, if libudev-dev is installed, why is it failing to find libudev.h?

mildsunrise, 2024-04-29:
wait... are you using nix? if so, libudev's development files may not be installed in the environment that nix is creating for you. please try to build the addon _outside_ of nix and see if it builds correctly.

mateja176, 2024-04-29:
> wait... are you using nix? if so, libudev's development files may not be installed in the environment that nix is creating for you. please try to build the addon _outside_ of nix and see if it builds correctly.

I’m simply using nix home-manager. Moreover, I’ve installed libudev-dev using apt. Where is the header’s file supposed to be located? p.s. I’ll try running the modified installation script.

mildsunrise, 2024-04-29:
> I’m simply using nix home-manager

in that case (AFAIU) yes, you're using nix, and the environment you get will be from nix, meaning libudev needs to be installed in the nix environment instead of in the outer system. (I've never used Nix so I could be wrong here)

a good way to know if the nix setup is the culprit here is to do `npm i usb` outside of the nix environment and see if it works there.

> Where is the header’s file supposed to be located?

it is often in `/usr/include/libudev.h`, but it really depends on your setup, i.e. whatever compiler you're using. the header should be in one of the system include directories of your compiler.

mateja176, 2024-04-30:
> it is often in `/usr/include/libudev.h`, but it really depends on your setup, i.e. whatever compiler you're using. the header should be in one of the system include directories of your compiler.

Interestingly, I was able to locate `/usr/include/libudev.h` on my system.
On the other hand, I'm not using [nix-shell](https://nixos.wiki/wiki/Development_environment_with_nix-shell) or any kind of virtual environment. However, I've only installed a `gcc`, `gnumake`, and python3 using nix home-manager. Do you think that's the reason why `libudev.h` cannot be found?

mildsunrise, 2024-04-30:
> I'm not using [nix-shell](https://nixos.wiki/wiki/Development_environment_with_nix-shell) or any kind of virtual environment

I'm not sure what you mean, but AFAIK home-manager will modify the environment upon session start, if only to override system commands / libs with the ones installed through nix.

> However, I've installed gcc, gnumake, and python3 using nix home-manager. Do you think that's the reason why libudev.h cannot be found?

Most likely. The header has to be in one of the directories that your compiler searches in. If your compiler was installed through nix, it would make sense that it is only searching in nix-controlled directories. As said, in this case libudev should be installed in the nix environment, not system-wise.

mateja176, 2024-05-02:
Even though apparently there is no nix package a equivalent of `libudev-dev` with the `libudev.h` file. I was able to install the `usb` after making the `/usr/bin/gcc` the default compiler. Appreciate your advice!

closed 2024-05-02 by mateja176

thegecko, 2024-05-02:
thanks @mildsunrise