node-usb/node-usb #780: Crash on Electron Quit on 2.13.0

semireg, 2024-06-24:
## Bug Description:

Seems to be a crash on quit.

```
FATAL ERROR:  
 1: 0x11381f4fc node::Buffer::New(v8::Isolate*, char*, unsigned long) [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
 2: 0x11381f5f0 node::OnFatalError(char const*, char const*) [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
 3: 0x11381f514 node::OnFatalError(char const*, char const*) [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
 4: 0x1137e17e8 napi_open_callback_scope [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
 5: 0x104b07c44 Napi::Error::Error(napi_env__*, napi_value__*) [/app/app/node_modules/usb/build/Release/usb_bindings.node]
 6: 0x104b14b3c handleCompletion(Transfer*) [/app/app/node_modules/usb/build/Release/usb_bindings.node]
 7: 0x104b0903c void Napi::details::WrapVoidCallback<Napi::ThreadSafeFunction::CallJS(napi_env__*, napi_value__*, void*, void*)::'lambda'()>(Napi::ThreadSafeFunction::CallJS(napi_env__*, napi_value__*, void*, void*)::'lambda'()) [/app/app/node_modules/usb/build/Release/usb_bindings.node]
 8: 0x104b08fe4 Napi::ThreadSafeFunction::CallJS(napi_env__*, napi_value__*, void*, void*) [/app/app/node_modules/usb/build/Release/usb_bindings.node]
 9: 0x1137e4c90 node_api_get_module_file_name [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
10: 0x10d3b8408 uv_async_send [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
11: 0x10d3c93c8 uv_free_interface_addresses [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
12: 0x10d3b884c uv_run [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
13: 0x1137c8338 node::EmitAsyncDestroy(node::Environment*, node::async_context) [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
14: 0x1137c89cc node::EmitAsyncDestroy(node::Environment*, node::async_context) [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
15: 0x11377ca4c node::FreeEnvironment(node::Environment*) [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
16: 0x10d4c42f0 v8::internal::HeapAllocator::HeapAllocator(v8::internal::Heap*) [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
17: 0x10d4ad3fc v8::internal::compiler::SLVerifierHintParametersOf(v8::internal::compiler::Operator const*) [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
18: 0x10f3f61cc v8::internal::SetupIsolateDelegate::SetupBuiltins(v8::internal::Isolate*, bool) [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
19: 0x10f3f7958 v8::internal::SetupIsolateDelegate::SetupBuiltins(v8::internal::Isolate*, bool) [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
20: 0x10f3f3c9c v8::internal::SetupIsolateDelegate::SetupBuiltins(v8::internal::Isolate*, bool) [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
21: 0x10d6a8648 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
22: 0x10d6a9564 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
23: 0x10d6a93b4 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
24: 0x10d6a7f34 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
25: 0x10d6a8044 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
26: 0x10d3c9c10 ElectronMain [/app/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
27: 0x1919de0e0 start [/usr/lib/dyld]
/app/node_modules/electron/dist/Electron.app/Contents/MacOS/Electron exited with signal SIGABRT
```

## Steps to Reproduce:

This is erratic and doesn't happen every time.

## Additional Information

- macOS Darwin Kernel Version 23.5.0 arm64
- Electron 26.6.9
- 2.13.0

thegecko, 2024-06-24:
libusb was updated in the latest release, can you confirm if the same crashes are seen with the previous release?

semireg, 2024-06-24:
I haven't studied the stack trace, but it does bare resemblance to the other crash I reported some time ago https://github.com/node-usb/node-usb/issues/560

thegecko, 2024-09-15:
Long shot: I wonder if this helps: https://github.com/node-usb/node-usb/pull/817