node-usb/node-usb #781: LIBUSB_ERROR_IO when writing to a reconnected device

tomek3e, 2024-06-27:
## Bug Description:
Getting LIBUSB_ERROR_IO upon trying to write to and endpoint of an open device which was previously connected, disconnect and reconnected.

## Steps to Reproduce:
1. Connect, device.open(), claim interface and call transfer write to a USB device. (works fine)
2. Detach the USB device.
3. Re-attach same USB device.
4. Open the re-connected device, claim interface and call transfer write to a USB device:
```
const device = findByIds(vid, pid);
device.open();
const interface = device.interfaces[0];
interface.claim();
interface.endpoints[0].transfer(data, (error)=> console.error(error?.message))
```
5. Transfer fails with callback error LIBUSB_ERROR_IO

## Additional Information
- Operating System: Windows 10
- Runtime Node 20, Node 18
- Node USB Version 2.11.0, 2.13.0

thegecko, 2024-06-28:
I wonder if the claimed interface is retained across the plugging?

tomek3e, 2024-07-01:
I tried calling `.release()` on it but there is an error in the callback.

thegecko, 2024-07-05:
This feels like a libusb error, but I lack time to investigate :(

thegecko, 2024-09-28:
I updated the test suite to run the open-claim-transfer tests both before and after a disconnection cycle:

https://github.com/node-usb/node-usb/tree/test-reconnected

It uses the same sequence you outline above.

I tried these tests on macos and windows 10 and saw no issue with my test device.

Do you have any other information?