node-usb/node-usb #795: unbantu environment node-usb in electron-builder A native build error occurred when executing the electron-builder --win --ia32 command

FgitYY, 2024-07-23:
ode v16.14.0
npm 8.3.1
"electron": "^20.2.0",
"electron-builder": "^23.3.3"
"usb": "^2.13.0"

The following error occurred:

cannot execute cause=exit status 1
errorOut=npm WARN using --force Recommended protections disabled.
npm ERR! code 1
npm ERR! path /project/code_master_test/web/node_modules/usb
npm ERR! command failed
npm ERR! command sh -c node-gyp-build
npm ERR! make: Entering directory '/project/code_master_test/web/node_modules/usb/build'
npm ERR! CC(target) Release/obj.target/libusb/libusb/libusb/core.o
npm ERR! CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
npm ERR! CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
npm ERR! CC(target) Release/obj.target/libusb/libusb/libusb/io.o
npm ERR! CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
npm ERR! CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
npm ERR! CC(target) Release/obj.target/libusb/libusb/libusb/os/events_posix.o
npm ERR! make: Leaving directory '/project/code_master_test/web/node_modules/usb/build'
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@10.2.0
npm ERR! gyp info using node@16.14.0 | linux | x64
npm ERR! gyp info find Python using Python version 3.8.10 found at "/usr/bin/python3"
npm ERR! gyp http GET https://electronjs.org/headers/v20.3.12/node-v20.3.12-headers.tar.gz
npm ERR! gyp http 200 https://artifacts.electronjs.org/headers/v20.3.12/node-v20.3.12-headers.tar.gz?force_headers_dist=1
npm ERR! gyp http GET https://electronjs.org/headers/v20.3.12/SHASUMS256.txt
npm ERR! gyp http 200 https://artifacts.electronjs.org/headers/v20.3.12/SHASUMS256.txt?force_headers_dist=1
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args '/project/code_master_test/web/node_modules/usb/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args 'binding.gyp',
npm ERR! gyp info spawn args '-f',
npm ERR! gyp info spawn args 'make',
npm ERR! gyp info spawn args '-I',
npm ERR! gyp info spawn args '/project/code_master_test/web/node_modules/usb/build/config.gypi',
npm ERR! gyp info spawn args '-I',
npm ERR! gyp info spawn args '/project/code_master_test/web/node_modules/usb/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args '-I',
npm ERR! gyp info spawn args '/home/nsfocus/.electron-gyp/20.3.12/include/node/common.gypi',
npm ERR! gyp info spawn args '-Dlibrary=shared_library',
npm ERR! gyp info spawn args '-Dvisibility=default',
npm ERR! gyp info spawn args '-Dnode_root_dir=/home/nsfocus/.electron-gyp/20.3.12',
npm ERR! gyp info spawn args '-Dnode_gyp_dir=/project/code_master_test/web/node_modules/usb/node_modules/node-gyp',
npm ERR! gyp info spawn args '-Dnode_lib_file=/home/nsfocus/.electron-gyp/20.3.12/<(target_arch)/node.lib',
npm ERR! gyp info spawn args '-Dmodule_root_dir=/project/code_master_test/web/node_modules/usb',
npm ERR! gyp info spawn args '-Dnode_engine=v8',
npm ERR! gyp info spawn args '--depth=.',
npm ERR! gyp info spawn args '--no-parallel',
npm ERR! gyp info spawn args '--generator-output',
npm ERR! gyp info spawn args 'build',
npm ERR! gyp info spawn args '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! In file included from /usr/include/bits/errno.h:26,
npm ERR! from /usr/include/errno.h:28,
npm ERR! from ../libusb/libusb/os/events_posix.c:23:
npm ERR! /usr/include/linux/errno.h:1:10: fatal error: asm/errno.h: No such file or directory
npm ERR! 1 | #include <asm/errno.h>
npm ERR! | ^~~~~~~~~~~~~
npm ERR! compilation terminated.
npm ERR! make: *** [libusb.target.mk:164: Release/obj.target/libusb/libusb/libusb/os/events_posix.o] Error 1
npm ERR! gyp ERR! build error
npm ERR! gyp ERR! stack Error: make failed with exit code: 2
npm ERR! gyp ERR! stack at ChildProcess. (/project/code_master_test/web/node_modules/usb/node_modules/node-gyp/lib/build.js:216:23)
npm ERR! gyp ERR! System Linux 3.10.0-1160.76.1.el7.x86_64
npm ERR! gyp ERR! command "/usr/local/bin/node" "/project/code_master_test/web/node_modules/usb/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /project/code_master_test/web/node_modules/usb
npm ERR! gyp ERR! node -v v16.14.0
npm ERR! gyp ERR! node-gyp -v v10.2.0
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/nsfocus/.npm/_logs/2024-07-19T09_53_40_857Z-debug-0.log

                command=/usr/local/bin/node /usr/local/lib/node_modules/npm/bin/npm-cli.js rebuild usb@2.13.0
                workingDir=

FgitYY, 2024-07-23:
I tried to install sudo apt-get install build-essential libudev-dev but I still got this error，Oddly, I was able to avoid this problem by using npm install usb@testing

FgitYY, 2024-07-23:
Current Ubuntu information
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.5 LTS
Release:        20.04
Codename:       focal

thegecko, 2024-07-23:
Node USB ships with all native modules precompiled for platforms it supports. Please suppress your build system from trying to build node USB again

FgitYY, 2024-07-23:
electron-builder Prevents the build system from trying to build the node USB again，Can you help me

FgitYY, 2024-07-23:
Yes, I solved this problem by adding configuration properties in electron-builder.yml. Add these three attributes:
buildDependenciesFromSource: true
nodeGypRebuild: false
npmRebuild: false

thegecko, 2024-09-15:
Thanks, I've added this to the readme

closed 2024-09-15 by thegecko