node-usb/node-usb #803: getDeviceList() does not update the list of devices properly

JonathanLeeIFX, 2024-08-28:
## Bug Description:
getDeviceList() will not show a device's new address if it is connected, removed, then reconnected.

## Steps to Reproduce:
I tested this using "usbview.exe" to check the Device Address and a Typescript program that tries to find a Silicon Labs C8051 device with  VID=0x10C4 and PID=0xEA61 every 3 seconds.
1. Connect the device and check the Device Address in "usbview.exe".
2. Launch the Typescript program. Observe that it can detect the device and reports the same Device Address.
3. Disconnect the device. Wait 2 seconds.
4. Reconnect the device. Observe that "usbview.exe" shows a different Device Address for the C8051, but the Typescript program continues to report the old Device Address.

## Additional Information
- Operating System: Windows 10
- Runtime: Node v16.17.0
- Node USB Version: usb@2.13.0

## Typescript to reproduce:

```
import * as usb from 'usb';
async function main() {
	let i = 0;
	do {
		await delay(3000);	
		const devices = usb.findByIds(<VENDOR_ID>, <PRODUCT_ID>);		
		console.log(`device run ${i}: `, devices);
		i++;
	} while (true);
}
function delay(ms: number) {
    return new Promise( resolve => setTimeout(resolve, ms) );
}
main();
```

thegecko, 2024-09-28:
Thanks for raising this and the reproducer code
I can confirm I see the same behaviour and the issue is apparent on windows 10. It looks like a race condition.
It doesn't seem to occur on macos, so seems seems to be OS specific.

I've had a play around with the code and even invalidating the internal cache doesn't fix the issue.

Breaking it down to the fewest calls with [libusb_get_device_list](https://github.com/node-usb/node-usb/blob/main/src/node_usb.cc#L104) and printing the address returned with [libusb_get_device_address](https://github.com/node-usb/node-usb/pull/821/files#diff-ba21e282fe0588b3a65e3ab4a5b27a48a13c68f2d4dfa7958567b90a4f96f5d1R110) still shows no change between disconnect/connect.

This leads me to believe its an issue with the `libusb` library on windows and needs to be logged [there](https://github.com/libusb/libusb/issues).

JonathanLeeIFX, 2024-10-02:
@thegecko After some discussion with **tormodvolden**, I've created an example project using C and the original libusb.dll: [libusbTest](https://github.com/JonathanLeeIFX/libusbTest).

I've included an example output from my program in the README. In the example, I started with my dongle connected, unplugged it after 3 loops, then plugged it back in at 7 loops. The device address updates correctly from `30` to `31`.

If I comment out line 66 `libusb_free_device_list(devs, 1);` in my example, I get the same behavior as we see when we run `GetDeviceList()`: the device disappears when I disconnect it, and retains its old address when I reconnect it. This leads me to believe that `libusb_free_device_list()` is not running properly for some reason.

thegecko, 2024-10-03:
Thanks for looking in to this. I was hoping your research uncovered a need to call `libusb_free_device_list`, but it appears [we already do](https://github.com/node-usb/node-usb/blob/main/src/node_usb.cc#L113).

I wonder if a reference to the device is being held somewhere...

goodwill, 2025-10-20:
Not sure if this is related, but for us using node-usb in linux docker container with privileged access, getDeviceList(), then disconnect a device (getDeviceList the device is gone), then reconnect the device, the list will not have the reconnected device. The only way to make this work apparently is to restart the node instance. This  is  really  not  great for  situation where  device might be removed and reconnected.