node-usb/node-usb #805: Windows 7 32-bit Report an error using nodeusb

FgitYY, 2024-09-02:
environment
node   v16.14.0
npm   8.3.1
"electron": "^20.2.0",
"electron-builder": "^23.3.3"
"usb": "2.4.3"

Windows 10 and 11, Windows 7 64-bit open without any problems, but Windows 7 32-bit open error:

[Window Title]
Error

[Main Instruction]
A JavaScript error occurred in the main process
A JavaScript error occurred in the main process
Uncaught Exception:TypeError: Cannot read properties of undefined (reading 'prototype’)at C:ProgramFiles\NsfocusVPNresourceslapp,asarnode_moduleshusb\distlusb\indexjs:22:38at Array,forEach (sandyymous>)
at Object.sanonymous>(C:ProgramFilesNsfocusVPN\resourceslapp,asarinode_moduleslusbidistlusblindexjs:21:63)at Module. compile (node:internal/modules/'cjs/loader:1120:14)
at Module, extensions..js (node:internal/modules/cjs/loader:1175:10)
at Module.load (node:internal/modules/cis/oader:988:32)at Module, load (node:internal/modules/cis/loader:829:12
at c, load (node:electron/js2c/asar bundle:5:13343)
at Module.require (nede:internal/modules/cjs/loader:1012:19)
at require (node:internal/modules/cjs/helpers:102:18)

thegecko, 2024-09-03:
Hmm, not much idea at all. Please ensure you have electron set up to not rebuild native packages

Gravityboom, 2024-11-04:
I encountered a similar error:

thegecko, 2024-11-06:
These look like different issues, however I'm aware of shipped products using both electron and Node-usb so this should work!

I think some version bisection is needed here as I'm really keen to find out when this started failing and whether this was due to a node-usb update or electron update.