node-usb/node-usb #836: \\?\C:\Users\EDY\Desktop\vxspace\node_modules\usb\prebuilds\win32-x64\node.napi.node

MF-institute, 2024-11-08:
错误应用程序名称: electron.exe，版本: 27.3.11.0，时间戳: 0x661ec0b6
错误模块名称: node.napi.node，版本: 0.0.0.0，时间戳: 0x66e75a59
异常代码: 0xc000041d
错误偏移量: 0x00000000000172e6
错误进程 ID: 0x0x3A94
错误应用程序启动时间: 0x0x1DB31B3FDA313BC
错误应用程序路径: C:\Users\EDY\Desktop\vxspace\node_modules\electron\dist\electron.exe
错误模块路径: \\?\C:\Users\EDY\Desktop\vxspace\node_modules\usb\prebuilds\win32-x64\node.napi.node
报告 ID: 8941cb3d-aacb-4489-bc6a-5ef09e6b4af5
错误程序包全名: 
错误程序包相对应用程序 ID:

thegecko, 2024-11-08:
Sorry, I don't understand this issue

MF-institute, 2024-11-11:
Using the usb plug and plug detection function, I need to replace the driver of a device after detecting a usb device, which then causes my application to crash. After checking the windows event, I find that the problem is under win32-x64\node.napi.node

MF-institute, 2024-11-11:
I want to know how to solve this problem

thegecko, 2024-11-11:
So you are switching out a driver out while using the library in electron? What is the error you see?