node-usb/node-usb #853: Exception during run: Error: Cannot find module

Martinfx, 2025-01-09:
## Bug Description:
Hi,
i wanted run test but i have problem: 

```
 mocha --require coffeescript/register --grep Module test/*

 Exception during run: Error: Cannot find module '/usr/home/maxfx/Documents/node-usb/dist/index.js'. Please verify that the package.json has a valid "main" entry
    at tryPackage (node:internal/modules/cjs/loader:494:19)
    at Function._findPath (node:internal/modules/cjs/loader:795:18)
    at Function._resolveFilename (node:internal/modules/cjs/loader:1235:27)
    at Function._load (node:internal/modules/cjs/loader:1075:27)
    at TracingChannel.traceSync (node:diagnostics_channel:315:14)
    at wrapModuleLoad (node:internal/modules/cjs/loader:218:24)
    at Module.require (node:internal/modules/cjs/loader:1340:12)
    at require (node:internal/modules/helpers:141:16)
    at Object.<anonymous> (/usr/home/maxfx/Documents/node-usb/test/usb.coffee:8:9)
    at Object.<anonymous> (/usr/home/maxfx/Documents/node-usb/test/usb.coffee:337:4)
    at Module._compile (node:internal/modules/cjs/loader:1546:14)
    at Object.loadFile (/usr/home/maxfx/Documents/node-usb/node_modules/coffeescript/lib/coffeescript/register.js:33:19)
    at Module.load (/usr/home/maxfx/Documents/node-usb/node_modules/coffeescript/lib/coffeescript/register.js:68:36)
    at Function._load (node:internal/modules/cjs/loader:1128:12)
    at TracingChannel.traceSync (node:diagnostics_channel:315:14)
    at wrapModuleLoad (node:internal/modules/cjs/loader:218:24)
    at Module.require (node:internal/modules/cjs/loader:1340:12)
    at require (node:internal/modules/helpers:141:16)
    at exports.requireOrImport (/usr/home/maxfx/Documents/node-usb/node_modules/mocha/lib/nodejs/esm-utils.js:53:16)
    at async exports.loadFilesAsync (/usr/home/maxfx/Documents/node-usb/node_modules/mocha/lib/nodejs/esm-utils.js:100:20)
    at async singleRun (/usr/home/maxfx/Documents/node-usb/node_modules/mocha/lib/cli/run-helpers.js:162:3)
    at async exports.handler (/usr/home/maxfx/Documents/node-usb/node_modules/mocha/lib/cli/run.js:375:5) {
  code: 'MODULE_NOT_FOUND',
  path: '/usr/home/maxfx/Documents/node-usb/package.json',
  requestPath: '../'
}
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

## Steps to Reproduce:
1. yarn test
2.
3.

## Additional Information
- Operating System: FreeBSD 14.2-RELEASE amd64
- Runtime: node - v22.11.0 , electron - v32.2.7
- Node USB Version (`npm ls usb`): usb@2.14.0

thegecko, 2025-01-09:
Please build the library first

Martinfx, 2025-01-09:
> Please build the library first

I build lib first.
1. yarn build
2. yarn test

Martinfx, 2025-01-09:
so yarn compile then yarn test. I am sorry.