node-usb/node-usb #857: Crash 'Symbol not found' when calling `detachKernelDriver()` on macOS

rubycon, 2025-02-03:
## Bug Description:
When calling the `detachKernelDriver()` method on a device interface on macOS node crashes with the following message :
```
dyld: lazy symbol binding failed: Symbol not found: _SecTaskCreateFromSelf
  Referenced from: <...>/node_modules/usb/prebuilds/darwin-x64+arm64/node.napi.node
  Expected in: flat namespace

dyld: Symbol not found: _SecTaskCreateFromSelf
  Referenced from: <...>/node_modules/usb/prebuilds/darwin-x64+arm64/node.napi.node
  Expected in: flat namespace

Abort trap: 6
```

Strangely `isKernelDriverActive()` and `attachKernelDriver()` do not crash and seems to work properly.

## Steps to Reproduce:
- Install `node-usb` with yarn or npm
- Run the following code :
```js
import usb from 'usb';

const device = usb.getDeviceList()[0];

device.open();

if (device.interfaces[0].isKernelDriverActive()) {
    device.interfaces[0].detachKernelDriver();
}
```

## Additional Information
- Operating System: macOS BigSur 11.6.1 (Apple M1)
- Runtime : Node 20.11.0
- Node USB Version: 2.14.0

thegecko, 2025-02-15:
According to this, Kernel driver detaching is not supported by libusb under macOS.

https://github.com/libusb/libusb/issues/908#issuecomment-820018553

rubycon, 2025-02-17:
Thanks for your reply @thegecko.

This issue is from 2021, since then the MacOS part of the FAQ has been updated :
> Updates in Feb 2022: with the merging of pull request #911, support for detaching kernel drivers with authorization has been added. This has been included in the upcoming 1.0.25 release. 

Looking at libusb code confirms that [attaching/detaching the kernel driver is in fact supported on MacOS](https://github.com/libusb/libusb/blob/d52e355daa09f17ce64819122cb067b8a2ee0d4b/libusb/os/darwin_usb.c#L2911). These methods are supported for MacOs 10.10 or higher but also depends on the Apple SDK version use [at build time](https://github.com/libusb/libusb/blob/d52e355daa09f17ce64819122cb067b8a2ee0d4b/libusb/os/darwin_usb.c#L110).

libusb now also provide a [`libusb_set_auto_detach_kernel_driver`](https://github.com/libusb/libusb/blob/master/libusb/core.c#L2197) method which could be nice if it was exposed through an init option in Node USB or even the default setting in the WebUSB API as it completely abstract the kernel driver.

thegecko, 2025-02-17:
Thanks for the updated info!

It should be quite easy to add that method and I agree it could be a good default.

thegecko, 2025-02-18:
I've added this in https://github.com/node-usb/node-usb/pull/859

Do you have any way of testing it?

rubycon, 2025-02-20:
Great @thegecko !

For now I can't build the package (on macOS BigSur 11.6.1 (Apple M1)). It fails with `No rule to make target `Release/obj.target/libusb/libusb/libusb/core.o', needed by `Release/usb.a'`. Can you help me with that ? Here's the full log :
```
gyp info it worked if it ends with ok
gyp info using node-gyp@10.1.0
gyp info using node@20.11.0 | darwin | arm64
gyp info find Python using Python version 3.13.2 found at "/opt/homebrew/opt/python@3.13/bin/python3.13"

gyp info spawn /opt/homebrew/opt/python@3.13/bin/python3.13
gyp info spawn args [
gyp info spawn args '/Users/username/node-usb/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args 'binding.gyp',
gyp info spawn args '-f',
gyp info spawn args 'make',
gyp info spawn args '-I',
gyp info spawn args '/Users/username/node-usb/build/config.gypi',
gyp info spawn args '-I',
gyp info spawn args '/Users/username/node-usb/node_modules/node-gyp/addon.gypi',
gyp info spawn args '-I',
gyp info spawn args '/Users/username/Library/Caches/node-gyp/20.11.0/include/node/common.gypi',
gyp info spawn args '-Dlibrary=shared_library',
gyp info spawn args '-Dvisibility=default',
gyp info spawn args '-Dnode_root_dir=/Users/username/Library/Caches/node-gyp/20.11.0',
gyp info spawn args '-Dnode_gyp_dir=/Users/username/node-usb/node_modules/node-gyp',
gyp info spawn args '-Dnode_lib_file=/Users/username/Library/Caches/node-gyp/20.11.0/<(target_arch)/node.lib',
gyp info spawn args '-Dmodule_root_dir=/Users/username/node-usb',
gyp info spawn args '-Dnode_engine=v8',
gyp info spawn args '--depth=.',
gyp info spawn args '--no-parallel',
gyp info spawn args '--generator-output',
gyp info spawn args 'build',
gyp info spawn args '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: *** No rule to make target `Release/obj.target/libusb/libusb/libusb/core.o', needed by `Release/usb.a'.  Stop.
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack at ChildProcess.<anonymous> (/Users/username/node-usb/node_modules/node-gyp/lib/build.js:209:23)
gyp ERR! System Darwin 20.6.0
gyp ERR! command "/Users/username/.nvm/versions/node/v20.11.0/bin/node" "/Users/username/node-usb/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/username/node-usb
gyp ERR! node -v v20.11.0
gyp ERR! node-gyp -v v10.1.0
gyp ERR! not ok
```

I see that you automatically set auto detach to `true` in node-usb regular API `open()` method. However I think it would be better not to, and instead set auto detach on `open()` in the WebUSB API only. The regular API needs to be more granular, the WebUSB is a (slightly) higher abstraction and a better fit for this kind auto setup.

rubycon, 2025-02-20:
@thegecko Sorry, I found how to build the package (clone with submodules > yarn install > yarn compile). I still had the same crash issue. Then I updated the MacOS target in `bindings.gyp` and `libusb.gypi` from 10.7 to 10.10 (as explained above). But now I have a different crash bug :
```
dyld: lazy symbol binding failed: Symbol not found: _libusb_init
  Referenced from: /Users/username/node-usb/build/Release/usb_bindings.node
  Expected in: flat namespace

dyld: Symbol not found: _libusb_init
  Referenced from: /Users/username/node-usb/build/Release/usb_bindings.node
  Expected in: flat namespace

Abort trap: 6
```
It happens when I import the package with node (`import usb from 'usb';`)

thegecko, 2025-02-20:
As this change is only in the TypeScript wrapper, node and libusb binaries remain the same.
You should be able to copy the .node files from the latest release on npm or simply run `yarn prebuild-download`.

I'll change the default setup as you suggest...

thegecko, 2025-02-20:
I just tried updating the MacOS deployment target (10.10) and it worked fine on my M3/Sonoma. The full test suite passed.

Node 10.18.1

Not sure where your crash is coming from :(

rubycon, 2025-02-21:
While investigating my failed attempt to build the package properly, it seems I found the cause of the original bug:
- `SecTaskCreateFromSelf()` is a method from Apple Security Framework;
- It is called once in `darwin_has_capture_entitlements()` (called only in `darwin_detach_kernel_driver()`)
- Security is not listed in Frameworks to linked in `OTHER_LDFLAGS` in `bindings.gyp`.

Adding it solved the initial issue for me, but then why you don't have this issue ? What are the step you take to build the package ? Also Node 10.18.1 seems really old, are you sure ?

thegecko, 2025-02-21:
Ooh, I get a popup on mac when I connect the device to authorise it (which I allow). Perhaps this means the entitlement then always passes.

Should we be adding security in OTHER_LDFLAGS?

rubycon, 2025-02-21:
```diff
 ['OS=="mac"', {
   'xcode_settings': {
     'OTHER_CFLAGS': [
       '-std=c++1y',
       '-stdlib=libc++',
       '-arch x86_64',
       '-arch arm64'
     ],
     'OTHER_LDFLAGS': [
       '-framework', 'CoreFoundation',
       '-framework', 'IOKit',
+      '-framework', 'Security',
       '-arch x86_64',
       '-arch arm64'
     ],
     'SDKROOT': 'macosx',
     'MACOSX_DEPLOYMENT_TARGET': '10.7'
   }
 }],
```

thegecko, 2025-02-21:
> Do you build libusb independently with XCode or with npm run install ?

`yarn install` or `yarn rebuild`

thegecko, 2025-02-21:
Added the flag to the test branch and checked that all tests still pass locally for me.

rubycon, 2025-02-21:
@thegecko Perfect! It works great here. I have 2 feedbacks : 
- It may not be necessary to bump MacOS minimal target version, has the support depends on the SDK version use on build. 10.7 seems fine and is more aligned with libusb target (MacOS >= 10.6)
- I need to run my code with sudo for the detach kernel driver method to work (if not I get the usual LIBUSB_ERROR_ACCESS error), it might be a good idea to mention necessary privilege for this method in the doc.

closed 2025-02-21 by thegecko

rubycon, 2025-02-21:
My pleasure. Thanks for the quick resolve!