node-usb/node-usb #869: Open fails with LIBUSB_ERROR_NOT_SUPPORTED on Windows as of 2.15.0

phidgets, 2025-04-10:
## Bug Description:
Opening a WebUSB device on Windows fails with LIBUSB_ERROR_NOT_SUPPORTED.

This is caused by webusb-device.js:58 -> this.device.setAutoDetachKernelDriver(this.autoDetachKernelDriver);

On Windows, libusb_set_auto_detach_kernel_driver is not supported, and always returns LIBUSB_ERROR_NOT_SUPPORTED.

## Steps to Reproduce:
1. get a list of devices with webusb.getDevices()
2. open the first device
3. LIBUSB_ERROR_NOT_SUPPORTED

## Additional Information
- Operating System: Windows 11
- Runtime (`node -v` or `electron -v`): v22.14.0
- Node USB Version (`npm ls usb`): 2.15.0

thegecko, 2025-07-10:
@phidgets can you confirm if #870 fixes your issue? I'll be honest, I don't know where I got on this!

thegecko, 2025-07-16:
Thanks, the fix is now released in v2.16.0